riscv/riscv-isa-manual #1928: Should PMA be implemented as readable CSRs or MMIO registers?

Steven-Li-Xiaogang, 2025-03-26:
[image]

1. I noticed that unprivileged spec mentioned "PMAs must also be readable by software to correctly access certain devices or to ....", so should we provide a way to software to get PMA settings. (custom CSRs or MMIO registers)
2. I found some existed implementation such as sifive U74, make PMA to be platform specific setting and can not be accessed by software (even read-only)

gfavor, 2025-03-26:
The spec does not specify how PMAs should be physically implemented.  So both implementation choices (as well as other choices) are allowable.

aswaterman, 2025-03-27:
Also, platforms might make PMAs accessible to software in by passing them in through a sideband (e.g. a devicetree in external NVM).  So implementations like the SiFive U74 you mentioned are capable of complying with this requirement.

Steven-Li-Xiaogang, 2025-03-27:
@aswaterman so how to explain this statement

> PMAs must also be readable by software to correctly access certain devices or to correctly configure other
hardware components that access memory, such as DMA engines.

ved-rivos, 2025-03-27:
Machine mode specific software can provide this information to supervisor mode software - that usually include device drivers - through means like device tree. Such machine mode specific software may read the PMA information from machine specific storage media such as a rom or soi flash or nvme. In some cases the memory regions may be dynamic - such as when machine mode specific software discovers dimms plugged into a machine and trains them or maps pcie memory into the system address space .

Steven-Li-Xiaogang, 2025-03-27:
@ved-rivos got it, thanks a lot. The spec statement makes me confused and thought that hardware must have a way to make software to obtain PMA setting, actually we can do a pure software way.

gfavor, 2025-03-27:
Closing since it sounds like the issue has been resolved.

closed 2025-03-27 by gfavor