riscv/riscv-isa-manual #1929: Should PMA be implemented as readable CSRs or MMIO registers?

Steven-Li-Xiaogang, 2025-03-26:
[image]

1. I noticed that unprivileged spec mentioned "PMAs must also be readable by software to correctly access certain devices or to ....", so should we provide a way to software to get PMA settings. (custom CSRs or MMIO registers)
2. I found some existed implementation such as sifive U74, make PMA to be platform specific setting and can not be accessed by software (even read-only)

gfavor, 2025-03-26:
The spec provides flexibility to "do the right thing" as far as what other attributs are effected by a PBMT override (past the specified overrides):

> Implementations may override additional PMAs not explicitly listed in Table 37. For example, ...

Steven-Li-Xiaogang, 2025-03-28:
@gfavor thanks a lot.

closed 2025-03-28 by Steven-Li-Xiaogang