riscv/riscv-isa-manual #1936: Questions about the behavior of sfence.w.inval and sfence.inval.ir.

NewPaulWalker, 2025-03-28:
I saw the following description in the Svinval extension：

> In typical usage, software will invalidate a range of virtual addresses in the addresstranslation caches by executing an SFENCE.W.INVAL instruction, executing a series of SINVAL.VMA, HINVAL.VVMA, or HINVAL.GVMA instructions to the addresses (and optionally ASIDs or VMIDs) in question, and then executing an SFENCE.INVAL.IR instruction.

And I have two questions:  

1. Can other instructions, such as computation or memory access instructions, be placed between `SFENCE.W.INVAL` and `SFENCE.INVAL.IR`? What should happen when a hart encounters such an instruction sequence?  

2. After executing several `SINVAL.VMA`, `HINVAL.VVMA`, or `HINVAL.GVMA` instructions following `SFENCE.W.INVAL`, is an `SFENCE.INVAL.IR` instruction always required?

gfavor, 2025-03-28:
1. Yes, that is allowed.  Translation of any memory accesses during that sequence are subject to the usual uncertainty as to which translation (among old and new ones) is used - since the PTE changes and associated invalidations aren't synchronized with following translations until the SFENCE.INVAL.IR is performed.

2. If the SFENCE.INVAL.IR is not performed, then there is no guaranteed full synchronization wrt translation of following loads/stores.

aswaterman, 2025-03-28:
In case anyone is wondering, failing to execute an SFENCE.INVAL.IR doesn't result in totally unpredictable behavior.  Each subsequent memory access will unpredictably use either the old translation or the new translation.  Other behaviors can't occur.

allenjbaum, 2025-03-28:
Yea, that's actually a useful distinction.

NewPaulWalker, 2025-03-30:
Thanks to @gfavor and @aswaterman for the detailed explanation and I will close this issue.

closed 2025-03-30 by NewPaulWalker