riscv/riscv-isa-manual #1938: Zcb instructions that map to I instructions and the forward progress guarantee

asb, 2025-03-31:
This is request for clarification.

The manual currently states:

> An LR/SC sequence begins with an LR instruction and ends with an SC instruction. The dynamic
code executed between the LR and SC instructions can only contain instructions from the base ''I''
instruction set, excluding loads, stores, backward jumps, taken backward branches, JALR, FENCE,
and SYSTEM instructions. If the ''C'' extension is supported, then compressed forms of the
aforementioned ''I'' instructions are also permitted.

Some of the Zcb are compressed forms of "I" instructions that aren't loads/stores/jumps:
* c.zext.b
* c.not

I would like to confirm if these instructions can be used within the lr/sc sequence without breaking the forward progress guarantee or not. As the wording hasn't been updated since Zcb I think it's become slightly confusing to interpret (given the existing of compressed forms of "I" instructions that aren't in "C"), so if the intent is that c.not/c.zext.b can't be used in this sequence I would propose rewording to:

> If the "C" extension is supported, then compressed forms of the aforementioned "I" instructions that are provided by the "C" extension are also permitted.

I note that both Clang/LLVM and GCC will produce c.not in some compiler-generated lr/sc sequences as it stands. We can look to fix this, but I would of course like to confirm that it was intended these instructions are excluded from the forward progress guarantee. (On the clang side we do a late stage expansion of atomics to carefully controlled sequences that match the forward progress guarantee, but in this case the final pre-emission compression of those I instructions introduces 'c.not'. I believe it's the same on the gcc/gas side).

Just as background, here is an example of current compiler behaviour whereby c.not is used:

```
$ cat t.c
char c;

void foo(void) {
  __sync_fetch_and_nand (&c, 1);
}
$ clang --target=riscv64-linux-gnu -march=rv64gc_zcb -O3 t.c -c -o t.clang.o
$ llvm-objdump -d -Mno-aliases t.clang.o

t.clang.o:	file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <foo>:
       0: 00000517     	auipc	a0, 0x0
       4: 00050513     	addi	a0, a0, 0x0
       8: ffc57593     	andi	a1, a0, -0x4
       c: 890d         	c.andi	a0, 0x3
       e: 050e         	c.slli	a0, 0x3
      10: 0ff00613     	addi	a2, zero, 0xff
      14: 00a6163b     	sllw	a2, a2, a0
      18: 4685         	c.li	a3, 0x1
      1a: 00a6953b     	sllw	a0, a3, a0
      1e: 1605a6af     	lr.w.aqrl	a3, (a1)
      22: 00a6f733     	and	a4, a3, a0
      26: 9f75         	c.not	a4
      28: 8f35         	c.xor	a4, a3
      2a: 8f71         	c.and	a4, a2
      2c: 8f35         	c.xor	a4, a3
      2e: 1ae5a72f     	sc.w.rl	a4, a4, (a1)
      32: e301         	c.bnez	a4, 0x32 <foo+0x32>
      34: 8082         	c.jr	ra
$ riscv64-linux-gnu-gcc -march=rv64gc_zcb -O3 t.c -c -o t.gcc.o
$ llvm-objdump -d -Mno-aliases t.gcc.o

t.gcc.o:	file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <foo>:
       0: 00000797     	auipc	a5, 0x0
       4: 00078793     	addi	a5, a5, 0x0
       8: 0037f693     	andi	a3, a5, 0x3
       c: 0036969b     	slliw	a3, a3, 0x3
      10: 0ff00713     	addi	a4, zero, 0xff
      14: 00d7173b     	sllw	a4, a4, a3
      18: 4605         	c.li	a2, 0x1
      1a: 00d616bb     	sllw	a3, a2, a3
      1e: 9bf1         	c.andi	a5, -0x4
      20: fff74613     	xori	a2, a4, -0x1

0000000000000024 <.L11>:
      24: 1607a5af     	lr.w.aqrl	a1, (a5)
      28: 00d5f533     	and	a0, a1, a3
      2c: 9d75         	c.not	a0
      2e: 8d79         	c.and	a0, a4
      30: 00c5f833     	and	a6, a1, a2
      34: 00a86833     	or	a6, a6, a0
      38: 1b07a52f     	sc.w.rl	a0, a6, (a5)
      3c: f565         	c.bnez	a0, 0x24 <.L11>
      3e: 8082         	c.jr	ra

```

aswaterman, 2025-03-31:
The intent is to allow `c.not` and `c.zext.b` to be used in constrained LR/SC loops, and the text supports this interpretation by using the more generic "compressed forms" language, rather than "instructions in the C extension".

As a separate issue, the text should say "Zca" instead of "C".  (This is just an editing error; we should've replaced "C" with "Zca" in most places when Zca was incorporated, but we've clearly missed a few.)

aswaterman, 2025-03-31:
Proposal is to address both issues by rewriting that sentence as follows.  (cc @gfavor @kasanovic)

```
If the Zca extension
is supported, then compressed forms of the aforementioned ''I'' 
instructions are also permitted, including those defined in extensions
other than Zca.
```

gfavor, 2025-04-01:
The proposal LGTM.

asb, 2025-04-01:
> Proposal is to address both issues by rewriting that sentence as follows. (cc [@gfavor](https://github.com/gfavor) [@kasanovic](https://github.com/kasanovic))
>
> ```
> If the Zca extension
> is supported, [...]

This seems fine to me. I suppose the practical impact of allowing _any_ compressed form of I instructions that are supported, is to effectively place the restriction that any future Zc or vendor X* extensions that define such instructions cannot be supported by trap-and-emulate on a system that has Zalrsc. Trap and emulate for such instructions would be pretty horrendous performance-wise, so it's hard to imagine someone wanting to do that outside of a fairly contrived case.

aswaterman, 2025-04-01:
That was my reasoning, as well.

allenjbaum, 2025-04-01:
If it is a contrived case, then there should be no objection to making that requirement explicit, right?
(e.g no op that could trap is permitted inside a constrained loop?) Of course, any constrained loop that crosses a page boundary could potentially trap, compressed or not.

allenjbaum, 2025-04-01:
Re-reading the spec to the end, trapping is allowed (in fact, that's one of the allowed conditions), so in theory trap&emulation shouldn't be an impediment.

closed 2025-04-01 by aswaterman

asb, 2025-04-02:
> Re-reading the spec to the end, trapping is allowed (in fact, that's one of the allowed conditions), so in theory trap&emulation shouldn't be an impediment.

Maybe we're talking past each other here, but I don't believe that's the case. The forward-progress guarantee requires:

> An LR/SC sequence begins with an LR instruction and ends with an SC instruction. The dynamic code executed between the LR and SC instructions can only contain instructions from the base ''I'' [...]

Implementing trap+emulate involves memory access, CSR access, `mret` so immediately breaks the condition.

Thanks Andrew and the ARC for a quick resolution.

allenjbaum, 2025-04-02:
I wasn't explaining myself clearly. Trapping is not allowed as guarantee
for success, but as one of the allowed conditions for forward progress.
That's related, but off-topic.

kasanovic, 2025-04-04:
These are unprivileged instructions so potentially made available in any privilege mode.  It is the responsibility of the execution environment to obey the forward-progress guarantee otherwise the EE has not implemented the instructions correctly.  The trap mentioned in the spec for constrained LR/SC loops is a visible trap (e.g., an interrupt) - not an invisible trap to support emulation.

commit b2421028 (2025-03-31): Clarify description of constrained LR/SC loops for Zca++

This PR fixes an editing error in which "C" should have been changed to
"Zca" when the various Zc extensions were incorporated.  It also clarifies
that any compressed instruction, not just those in Zca, that map to one of the
permitted "I" instructions can be used in a constrained LR/SC loop.  This
was always the case, but it was stated obliquely.

Resolves #1938
commit 2c78a48f (2025-04-01): Clarify description of constrained LR/SC loops for Zca++

This PR fixes an editing error in which "C" should have been changed to
"Zca" when the various Zc extensions were incorporated.  It also clarifies
that any compressed instruction, not just those in Zca, that map to one of the
permitted "I" instructions can be used in a constrained LR/SC loop.  This
was always the case, but it was stated obliquely.

Resolves #1938
PR #1939 (2025-04-01): Clarify description of constrained LR/SC loops for Zca++
This PR fixes an editing error in which "C" should have been changed to "Zca" when the various Zc extensions were incorporated.  It also clarifies that any compressed instruction, not just those in Zca, that map to one of the permitted "I" instructions can be used in a constrained LR/SC loop.  This was always the case, but it was stated obliquely.

Resolves #1938
commit 372edf99 (2025-04-01): Clarify description of constrained LR/SC loops for Zca++ (#1939)

This PR fixes an editing error in which "C" should have been changed to
"Zca" when the various Zc extensions were incorporated.  It also clarifies
that any compressed instruction, not just those in Zca, that map to one of the
permitted "I" instructions can be used in a constrained LR/SC loop.  This
was always the case, but it was stated obliquely.

Resolves #1938