riscv/riscv-isa-manual #1944: the order of virtual address checking and ignore translation need to be clarified

qin996, 2025-04-02:
In pointer masking spec, it describes as follow.

> transformed_effective_address = {{PMLEN{effective_address[XLEN-PMLEN-1]}}, effective_address[XLEN-PMLEN-1:0]}

> For a virtual address to be valid, all bits in the unused portion of the address must be the same as the Most Significant Bit (MSB) of the used portion. For example, when page-based 48-bit virtual memory (Sv48) is used, load/store effective addresses, which are 64 bits, must have bits 63–48 all set to bit 47, or else a page-fault exception will occur.

so the bits 63-48 of effective address must equal to bit 47 for sv48. And the ignore transformation is defined as a transformation of the effective address. any virtual memory checks or translation steps apply to the already-transformed address according to https://github.com/riscv/riscv-j-extension/issues/92. Due to the tag information stored in the high bits of address, in the specific implementation, ignore translation is performed first, and then the 63-48 bits of the virtual address are checked to see if they are all zeros or all ones. however, according to the formula for ignore translation, use effective_address, so it may require all zero or all one check of the virtual address first. So does the order of virtual address checking and ignore translation need to be clarified?

for example, va = 0x12ff_ff81_1300_0000 and PMLEN=7, sv48, When performing an ignore translation, is it right to choose `transformerd_efficienta_address= {{PMLEN{VA[47]}},VA[47:0]}`?
if satp.mode = bare, `transformed_effective_address = {{PMLEN{0}}, PA[47:0]}` for physical addresses.

gfavor, 2025-04-02:
Even though this is a question about an arch extension in the ISA manuals, I strongly suggest posting this question to the J group (tech-j-ext@lists.risc-v.org).  That is where the right people to respond to this, will see this question.

As far as not responding to a prior question, if it is the one that was raised over the past week or two, the J group actually discussed that question in a TG meeting and provided a resolution on their email list and (I believe) in their meeting minutes.  If it was a different question, then raise the question again on the email list, but also address the chair of the TG as well (Martin Maas <mmaas@google.com>).  In all cases, the people that developed the pointer masking extension and are in a position to accurately answer questions, are on that email list.

qin996, 2025-04-02:
thanks, but I have ask the same question in J extension by `issue`, no one has replied yet. Can you help push it forward?

qin996, 2025-04-02:
Yes, the previous one has already found the answer; but for the current title of `the order of virtual address checking and ignore translation need to be clarified`, no one replied. https://github.com/riscv/riscv-j-extension/issues/99

radimkrcmar, 2025-04-02:
> So does the order of virtual address checking and ignore translation need to be clarified?

I think it is clear, ignore transformation first.  Quoting ISM II: 24.2.2. The “Ignore” Transformation:

```
it is semantically equivalent to replacing a subset of instructions (e.g., loads and stores) with an
instruction sequence that applies the ignore operation to the target address of this instruction
and then applies the instruction to the transformed address. 
```

Your examples have a small mistake as you forgot the bits [56:48],
> for example, va = 0x12ff_ff81_1300_0000 and PMLEN=7, sv48, When performing an ignore translation, is it right to choose transformerd_efficienta_address= {{PMLEN{VA[47]}},VA[47:0]}?
```
For virtual addresses, it replaces the upper PMLEN bits with the sign extension of the PMLEN+1st bit.
```

Should be {{PMLEN{VA[56]}}, VA[56:0]}.

> if satp.mode = bare, transformed_effective_address = {{PMLEN{0}}, PA[47:0]} for physical addresses.
```
When applied to a physical address, the ignore transformation replaces the upper PMLEN bits with 0. 
```
{{PMLEN{0}}, PA[56:0]}

gfavor, 2025-04-05:
Closing as this issue appears settled.

closed 2025-04-05 by gfavor