riscv/riscv-isa-manual #1952: Is this a typo in machine.adoc’s nested traps description?

Jrvvv, 2025-04-04:
Hi!

The specification states in the Machine Mode description of mstatus:

>  Interrupts for lower-privilege modes, w<x, are always globally disabled regardless of the setting of any global wIE bit for the lower-privilege mode. Interrupts for higher-privilege modes, y>x, are always globally enabled regardless of the setting of the global yIE bit for the higher-privilege mode.

According to this, y denotes a higher-privilege mode than x.

However, the next example about nested interrupts uses the same symbols but describes the opposite scenario:

> xPIE holds the value of the interrupt-enable bit active prior to the trap, and xPP holds the previous privilege mode. The xPP fields can only hold privilege modes up to x, so MPP is two bits wide and SPP is one bit wide. **When a trap is taken from privilege mode y into privilege mode x**, xPIE is set to the value of xIE; xIE is set to 0; and xPP is set to y. 

As I understand it, this example describes a transition from a lower-privilege mode to a higher-privilege mode (e.g., from User to Supervisor or from Supervisor to Machine mode). But since the same symbols (y > x) are used, it incorrectly implies that the trap goes from Machine (y) to Supervisor (x)— which contradicts standard RISC-V behavior.

Am I misunderstanding something, or is this a typo in the specification?

Thanks!

gfavor, 2025-04-05:
The two paragraphs should be read separately of each other. 'y' represents a higher priv mode than 'x' in the first paragraph, and 'y' implicitly represents an equal or lower priv mode than 'x' in the second paragraph.  (Simply replacing 'y' by 'w' in the second paragraph would not be correct.)

gfavor, 2025-04-09:
Closing since it has been several days with no further questions.