riscv/riscv-isa-manual #1957: question of Subtraction Wrap-Around in Vector vasubu.vv Instruction

pychego, 2025-04-08:
I am currently verifying the vasubu.vv instruction, with SEW = 8,  no inactive element. One element of vs2 is 0x6F, and the corresponding element in vs1 is 0xD9. I have a question. According to the vector spec, the subtraction result will ignore overflow and wrap around. My understanding is that 0x6F - 0xD9 = 0xFFFFFF96. After ignoring the overflow in the subtraction, the result is 0x96, which is an unsigned number. Then, shifting it results in 0b01001011, which is 0x4b. However, according to the calculation by Spike, the shifting includes the sign bit in front, resulting in 0b11001011, which is 0xcb. I think this does not conform to the meaning of the spec. Could you please explain which understanding is correct?

The relevant description in the spec is as follows:

[image]

[image]

[image]

pychego, 2025-04-08:
中文版本:  我现在在验证vasubu.vv指令, SEW=8, 其中vs2的一个元素为0x6F, vs1中对应的元素为0xD9, 我现在有一个疑问, 按照vector spec中所说, 减法结果将忽略溢出, 并产生回绕, 我的理解是0x6F - 0xD9 = 0xFFFFFF96, 减法忽略溢出后为0x96, 为一个无符号数, 然后进行移位得到0b01001011, 即0x4b; 但是根据spike计算这个减法, 得到的移位是带上了前面的符号位进行移位, 结算结果为0b11001011, 即0xcb,我认为这个和spec的含义不符合, 请问可以解答一下哪种理解是正确的吗?

gfavor, 2025-04-08:
By default the arch spec is correct and Spike would be suspect.  So this question should be opened against riscv-isa-sim (and closed here).

topperc, 2025-04-08:
If I'm reading the qemu code right it would appear to agrees with Spike. The subtraction and shift are done in larger precision and then truncate to the SEW. https://github.com/qemu/qemu/blob/dfaecc04c46d298e9ee81bd0ca96d8754f1c27ed/target/riscv/vector_helper.c#L2696

The sail code here also does the subtraction and shift using a width of SEW+1. https://github.com/riscv/sail-riscv/blob/3dd1e5954223e9c2c1e6f28274be8a0128043f5e/model/riscv_insts_vext_arith.sail#L1299

aswaterman, 2025-04-08:
The answer is essentially as @topperc said: the arithmetic is performed in infinite precision, then truncated to SEW bits as a final step.  The error in your computation is that you truncated the difference to SEW bits immediately after the subtraction.  Since bit 8 of the infinite-precision difference is 1, bit 7 of the shifted result becomes 1.  (We don't need to confuse ourselves by talking about sign bits here; this is all true in unsigned arithmetic.)

Perhaps the spec could be a little clearer, but here's an explanation for why my interpretation is the only valid one.  How could the related averaging-add instruction `vaaddu` possibly work if we truncated to SEW bits before shifting?  Suppose we want to average 200 and 220; we know the answer should be 210.  If we truncated before shifting, we'd obtain the result 82, which is clearly incorrect.

gfavor, 2025-04-08:
@aswaterman Thanks for stepping in with a "golden" answer.

pychego, 2025-04-09:
Thanks for your help!

aswaterman, 2025-04-09:
You're welcome!

closed 2025-04-09 by aswaterman

aswaterman, 2025-04-09:
I added a clarifying remark in the spec; see #1960

PR #1960 (2025-04-10): Clarify that vaadd intermediate results aren't truncated
Resolves #1957