riscv/riscv-isa-manual #1958: Implementing floating-point min/max reductions as a tree

awiseAtCodasip, 2025-04-08:
Unordered single-precision floating-point sum reduction allows the following latitude for implementations:

> Note: As a consequence of this definition, implementations need not propagate NaN payloads through the reduction tree when no elements are active. In particular, if no elements are active and the scalar input is NaN, implementations are permitted to canonicalize the NaN and, if the NaN is signaling, set the invalid exception flag. Implementations are alternatively permitted to pass through the original NaN and set no exception flags, as with vfredosum.

There is no equivalent for single-width floating-point max and min reductions which, despite it being noted that they should "return the same final value and raise the same exception flags regardless of operation order" and are therefore inherently unordered operations, are required to:

> Note: If no elements are active, the scalar in vs1[0] is simply copied to the destination register, without canonicalizing NaN values and without setting any exception flags.

This therefore makes it difficult to implement `vfredmin.vs` and `vfredmax.vs` as a tree reduction of the vector elements in `vs2` by replacing inactive elements by a suitable "identity" element. For any choice of bit pattern to use as the identity element it is impossible to distinguish, at the input to the min/max operation that deals with the scalar, `vs1[0]`, whether the identity is present as a result of there being no active vector elements or there being some active elements that just happen to have this identity element value.

Is this intentional?

Could the following alternative wording for the note be substituted?

NOTE: If no elements are active and the scalar input is NaN, implementations are permitted to canonicalize the NaN and, if the NaN is signaling, set the invalid exception flag. Implementations are alternatively permitted to pass through the original NaN and set no exception flags.

gfavor, 2025-04-08:
Since this question is about the why behind the arch spec, it will be better to raise this question on the vector SIG email list (where there are people that were involved with RVV development).

closed 2025-04-08 by gfavor

aswaterman, 2025-04-08:
@gfavor Agreed in general, but I can answer this one since I stumbled across it and am responsible for this part of the spec.

The current behavior is deliberate because it is equivalent to a serial application of the FMIN instruction and is therefore targetable in more autovectorization scenarios.  What you propose amounts to a spec change, which, in addition to the usual problems associated with a spec change, would axe the property I mentioned.  I know it is possible to make a conforming implementation using a reduction tree of a different shape with minimal additional hardware cost, because I've done it--admittedly with some annoyance.

gfavor, 2025-04-08:
@aswaterman Thanks for stepping in and short-circuiting the matter.