riscv/riscv-isa-manual #1961: Could vector instruction load device region data beyond it is need？

AlexGJL, 2025-04-09:
Could vector instruction load device region data more than it need？for example:
  1. A vector instruction aims to load 16B data，but consider the mask register value, the bytemask is 16'h5555 ,  so byte 16'haaaa is not necessary, could the vector load instruction still access 16B data in device(I/)) region?
  2. A vector instruction aims to load 14B data, the bytemask is 16'h3fff, could the vector load instruction still access 16B data in device(I/)) region?

gfavor, 2025-04-09:
If the access is to a PMA I/O region (versus a PMA main memory region), then only non-speculative access is allowed.  And as the RVV spec says:

> Non-idempotent memory regions must not have been updated for indices equal to or greater than the element that caused a synchronous trap during a vector store instruction.

So either the whole vector acess is successfully performed, only only access to the elements up to the faulting element.

closed 2025-04-09 by gfavor

AlexGJL, 2025-04-09:
Thank you for your reply about exception deal.  how about the I/O region data access rule, Could vector instruction load device region data more than it need？

AlexGJL, 2025-04-09:
Hi aswaterman，could you help resolve this  issue，thank you very much， @aswaterman

gfavor, 2025-04-09:
My prior post effectively answers your question.  Since PMA I/O regions typically are non-idempotent (i.e. disallowing speculative accesses), vector load accesses to such "device" regions cannot access more than "is needed".

AlexGJL, 2025-04-10:
Ok，thank you very much. arm allows accessing device region more than "is needed", risc-v is different at this point.