riscv/riscv-isa-manual #1962: Waiting for MIP.MSIP bit to be cleared after handling a software interrupt

bantierr, 2025-04-09:
Sorry to reopen, but I don’t think this answered my previous question [#1955 ](https://github.com/riscv/riscv-isa-manual/issues/1955)

The section quoted as an answer explains that there might be an undefined period between the time where the pending bit in mip rises/falls and the time where the conditions for a trap are evaluated.
I am asking how long it can take for MSIP to fall/rise after issuing the store instruction to the memory mapped register that will clear MSIP.

Thank you for the answer

aswaterman, 2025-04-09:
The spec says all that it can say.  If you know for sure that a transition is coming, you can poll for it.  Normal system software just tolerates that these events have arbitrary propagation delay, while optimizing for the common case that the delay is very short.

bantierr, 2025-04-09:
Thanks for the quick answer !

closed 2025-04-09 by bantierr