riscv/riscv-isa-manual #1969: For a misaligned load/store that is split into two accesses due to crossing page boundaries, if one of the addresses has the NC (non-cacheable) attribute and the other has the DEV (device) attribute, what type of fault should be generated?

AlexGJL, 2025-04-11:
If a misaligned load/store access is split into two accesses due to crossing page boundaries, and one of the split accesses returns an address with device attributes, should this trigger a page fault or a misaligned fault?

gfavor, 2025-04-11:
Definitely not a page fault since this isn't an MMU-related violation.  This boils down to PMAs (and any overrides by Svpbmt).  If the PMA region that has device attributes includes requiring that accesses be aligned, then a violation of that PMA requirement would result in an Access Fault.

But raising an address-misaligned exception is also a valid possibility - as the spec suggests: "Implementations may raise access-fault exceptions instead of address-misaligned exceptions for some misaligned accesses, indicating the instruction should not be emulated by a trap handler".

In short, if the misaligned access should be emulated, then raise a address-misaligned exception, else raise an access fault exception.  (Or, if misaligned mixed-attribute cases like this are supported by the implementation, then no exception would be raised.)

Steven-Li-Xiaogang, 2025-04-14:
@gfavor 

> In short, if the misaligned access should be emulated, then raise a address-misaligned exception, else raise an access fault exception. (Or, if misaligned mixed-attribute cases like this are supported by the implementation, then no exception would be raised.)

How to explain the misaligned access should be emulated, a misaligned access with hardware support? if hardware does not support misaligned access, then software could not emulate a misaligned access, an acces-fault exception should be raised?

aswaterman, 2025-04-15:
It's a platform or implementation choice, but normally it is undesirable to emulate misaligned accesses to side-effect-ful regions.

gfavor, 2025-04-16:
Closing since it looks like there isn't any further discussion to be had.