riscv/riscv-isa-manual #1972: Inconsistent read-only zero behavior in mstatus virtual memory-related bits

davidharrishmc, 2025-04-11:
machine.adoc says

SUM is read-only 0 if S-mode is not supported or if satp.MODE is read-only 0.
MXR is read-only 0 if S-mode is not supported.
TVM is read-only 0 when S-mode is not supported.

All three of these bits only affect behavior when virtual memory is active (or an instruction is issued that affects virtual memory).  For consistency, should they all be read-only 0 if sat.MODE is read-only 0?

gfavor, 2025-04-12:
Irrespective of whether it would be more appropriate or less appropriate  for MXR and TVM to be read-only zero when satp.MODE is read-only zero, the current long ratified spec for these is that they remain writable (but ignored) when satp.MODE is read-only zero.  Changing this would be a backward incompatible change.  And in any case changing a ratified spec is not allowed.

aswaterman, 2025-04-12:
The ratified-spec issue is a good enough reason on its own, but there's also a hypothetical technical justification.

In at least one of the various SPMP proposals, we proposed to require execution of an `sfence.vma x0, x0` instruction to synchronize with respect to SPMP register modifications, even if virtual memory is not supported.  If that proposal is adopted, then TVM will be of utility when virtualizing those systems.

MXR also plays a role in at least one SPMP proposal, performing the same function it does when virtual memory is enabled.

So, while these fields are vestigial in the current bare-S mode, soon they likely won't be.

davidharrishmc, 2025-04-12:
Thank you for the information.

closed 2025-04-12 by davidharrishmc