riscv/riscv-isa-manual #1973: In SVADU extension, memory disambiguation between PTE.D update and subsequent load

haosun86, 2025-04-12:
Hi,

In SVADU extension, could you please help confirm whether the following scenario can exist?
_Older store involves hardware update PTE.D bit, and younger load read same address of that PTE?_

If this scenario exists, does hardware need to maintain the order between PTE.D update and that younger load?

Looking forwarding to your reply!
Thanks
Hao Sun

aswaterman, 2025-04-12:
Yes, the spec explicitly mandates that a younger load will observe the D bit having been updated: "updates to the D bit [...] must be [...] observed in program order by the local hart".

https://github.com/riscv/riscv-isa-manual/blob/567502bebd29d3f9d9a3b4cd440287ffcfdc8c40/src/supervisor.adoc#L1441-L1443

haosun86, 2025-04-12:
Thanks so much

aswaterman, 2025-04-12:
You're welcome!

closed 2025-04-12 by aswaterman