riscv/riscv-isa-manual #1976: Dose vec instruction nf argument affect the element size？

AlexGJL, 2025-04-17:
1.spec require that vector memory instruction should naturally aligned to the size of the element, either the element is transferred successfully or an address misaligned exception is raised on that element，the element size is eew or （nf+1）*eew for segement load/store ?
2.vec Unit-stride Fault-Only-First Loads instruction would detect the exception of element0 , dose elemnet0 include (nf+1）*eew？

aswaterman, 2025-04-17:
1. No, each element within a segment is EEW bits wide, so the total size of a segment is (NFIELDS * EEW) bits.

2. Yes, if an exception is detected partway through accessing the zeroth segment, regardless of which element within that segment, the trap is taken.

AlexGJL, 2025-04-17:
thank you for your reply,  for the first answer, 1. index load store use sew for value, is sew is its element size?; 2. once addr does't align to  (NFIELDS * EEW)  would raise an exception for segment load/store?

aswaterman, 2025-04-17:
1. Yes, SEW supplies the data elements' EEW.
2. No, same alignment requirements as usual (some implementations might require EEW alignment).  (Considering that NFIELDS might not be a power of 2, (NFIELDS * EEW) alignment doesn't make sense anyway.)

closed 2025-04-18 by AlexGJL