riscv/riscv-isa-manual #1977: Questions about G-Stage PTE AD update

vicgithubu, 2025-04-17:
Considering the relaxation of G-Stage PTE D bit update(issue #1665), the following scenario may happen during two-stage translation:
1. During VS-Stage Level i implicit read, G-Stage Leaf PTE is found, A=1, D = 0 (assume menvcfg.ADUE=1 here)
2. Try to update G Stage PTE D-bit speculatively, but encountered pmp.W=0 or bus error 
3. then read VS-Stage Level i PTE, and found that:
   3.a VS-Stage Level i PTE is not a Leaf
   3.b VS-Stage Level i PTE is a Leaf, and AD = 2’b11, no need to update AD bit 
   3.c VS-Stage Level i PTE is a Leaf, and AD != 2’b11,  and need to update A/D bit according to original inst type

Question 1: Should an access fault be raised in step 2, 3.a or 3.b or 3.c?

Another question is about G-Stage PTE A bit check under this scenario:
During VS-Stage Level i implicit read, G-Stage Leaf PTE is found, A = 0.

Question 2: If menvcfg.ADUE=0 now, should a guest page fault be raised?
(or should VS implicit read also check G-leaf A bit?)

The 3rd question is about G-Stage PTE A bit update under similar scenario as Question 1:
1. During VS-Stage Level i implicit read, G-Stage Leaf PTE found, A=0 (assume menvcfg.ADUE=1 here)
2. Update G Stage PTE A-bit, but encountered pmp.W=0 or bus error 

Question 3: Should an access fault be raised in step 2?

Please help to confirm these questions, THANKS!

ved-rivos, 2025-04-17:
Q1: If G-stage PTE has W=1, then a D bit update may be attempted and lead to an access fault exception being raised. If G-stage PTE has W=0, then no D bit update is attempted and 3c will raise a guest page fault exception.
Q2: Guest page fault exception is raised.
Q3: Attempt to set G-stage PTE A-bit raises an access fault exception.

vicgithubu, 2025-04-19:
Thanks a lot!

closed 2025-04-19 by vicgithubu