riscv/riscv-isa-manual #1985: Which Exception needs to be raised for STIMECMP/VSTIMECMP CSR access when STCE in menvcfg is one and STCE in henvcfg is zero

harsha-gunturi, 2025-04-24:
Hi,

As per Priv Spec,

When STCE in menvcfg is one and STCE in henvcfg is zero, an attempt to access stimecmp (really vstimecmp) when V = 1 raises a virtual instruction exception, and VSTIP in hip reverts to its defined behavior as if this extension is not implemented.

Also, from below reference for when Virtual Exception should be raised,
When V=1, a virtual-instruction exception (code 22) is normally raised instead of an illegal-instruction exception if the attempted instruction is HS qualified but is prevented from executing when V=1 either due to insufficient privilege or because the instruction is expressly disabled by a supervisor or hypervisor CSR such as scounteren or hcounteren.

Accessing STIMECMP in VS mode when STCE in menvcfg is one, makes it HS-qualified but this access has sufficient privilege. 
STIMECMP can be accessed in VS mode, permissions are sufficient, and the above rule is not applicable.

1. Accessing STIMECMP CSR in VS mode when menvcfg is one and STCE in henvcfg is zero, should raise an illegal instruction exception.
2. Accessing STIMECMP CSR in VU mode when menvcfg is one and STCE in henvcfg is zero, should raise a virual instruction exception.
3. Accessing VSTIMECMP CSR in VS/VU mode when menvcfg is one and STCE in henvcfg is zero, should raise a virual instruction exception.

Could you address the above statements.

gfavor, 2025-04-26:
'stimecmp' is HS-qualified, but not because of "accessing STIMECMP in VS-mode ...".  The spec says that "An instruction is HS-qualified if it would be valid to execute in HS-mode".

And 'stimecmp' is prevented from executing due to insufficient privilege in VU-mode (just as in U-mode), but not in VS-mode (just as in S/HS mode).  And 'stimecmp' is prevented from executing due to being expressly disabled by the henvcfg.

Also note that the spec says: " When V=1, an attempt to read or write a VS CSR directly by its own separate CSR address causes a virtual-instruction exception. (Attempts from U-mode cause an illegal-instruction exception as usual.) "

So I believe these statements imply 'No' to question 1, 'Yes' to question 2, and 'Yes and No' to question 3.

/cc @jhauser-us to confirm or deny the preceding answers.

harsha-gunturi, 2025-04-26:
@gfavor, Thanks for the quick response.

Qustion 3, Accessing VSTIMECMP CSR in VS/VU mode when menvcfg is one and STCE in henvcfg is zero, should raise a virual instruction exception.
--> Yes and No to this question.
Yes, to Accessing VSTIMECMP CSR in VS mode when menvcfg is one and STCE in henvcfg is zero, should raise a virual instruction exception.
No, to Accessing VSTIMECMP CSR in VU mode when menvcfg is one and STCE in henvcfg is zero, should raise a virual instruction exception.
As per spec "When V=1, an attempt to read or write a VS CSR directly by its own separate CSR address causes a virtual-instruction exception" 
So accessing VSTIMECMP in VU mode should also result in Virtual Instruction Exception.

Question 1, Accessing STIMECMP CSR in VS mode when menvcfg is one and STCE in henvcfg is zero, should raise an illegal instruction exception.
STIMECMP access in VS mode can be executed and this is HS-qualified, so we can say It has sufficient privilege?
If it has sufficient privilege then illegal exception should be raised instead on virtual, as spec says "When V=1, a virtual-instruction exception (code 22) is normally raised instead of an illegal-instruction exception if the attempted instruction is HS qualified but is prevented from executing when V=1 either due to **insufficient privilege** or because the instruction is expressly disabled by a supervisor or hypervisor CSR such as scounteren or hcounteren."

Correct me if my understanding is wrong.

gfavor, 2025-04-26:
Re question 3, the spec says it all:  "When V=1, an attempt to read or write a VS CSR directly by its own separate CSR address causes a virtual-instruction exception. (Attempts from U-mode cause an illegal-instruction exception as usual.)"  So virtual-instruction exception in VS-mode, and illegal-instruction exception in VU-mode.  (And yes, the spec text could stand to be more precise.  But the parenthetical statement is in the context of the preceding sentence that it is clarifying - as well as in the context of the whole paragraph that is about V=1 behavior.)

Re question 1 (and in general), the spec says "When V=1, a virtual-instruction exception is normally raised ... if the attempted instruction is HS qualified but is prevented from executing when V=1 either due to insufficient privilege **or because the instruction is expressly disabled by a supervisor or hypervisor CSR** ...".  The bold clause applies as noted in my last post.  So a virtual-instruction exception is raised.

jhauser-us, 2025-04-26:
Statements 1, 2, and 3 at the top are false, true, and true.  All of the listed cases should raise a virtual-instruction exception.  One reason applies to statements 1 and 2 (an HS-qualified instruction, but prevented by `henvcfg`), while a different reason applies to statement 3 (an attempt to access a VS CSR directly).

@gfavor wrote:
> So virtual-instruction exception in VS-mode, and illegal-instruction exception in VU-mode.

Not quite.  @harsha-gunturi was correct:  A virtual-instruction exception should be rasied in both VS-mode and VU-mode.  Both VS- and VU-modes have virtualization mode V = 1.  The spec says an illegal-instruction exception is raised for "attempts from U-mode", but that isn't intended to include VU-mode.

jhauser-us, 2025-04-26:
I wrote:
> One reason applies to statements 1 and 2 (an HS-qualified instruction, but prevented by `henvcfg`), while a different reason applies to statement 3 (an attempt to access a VS CSR directly).

Correction:  Here are the reasons for each case:
1. an HS-qualified instruction, but prevented by `henvcfg`.
2. an HS-qualified instruction, but VU-mode has insufficient privilege.
3. an attempt to access a VS CSR directly when V = 1.

harsha-gunturi, 2025-04-27:
Thanks for clarifying and quick responses.

closed 2025-04-27 by harsha-gunturi