riscv/riscv-isa-manual #1989: When the vsetvl, vsetvli, or vsetivli instructions set the vill bit, must all other bits in vtype and the vl register be set to 0?

shengqilong, 2025-04-27:
```
When rs1=x0 and rd=x0, the instruction operates as if the current vector length in vl is used as the AVL, and the
resulting value is written to vl, but not to a destination register. This form can only be used when VLMAX and hence
vl is not actually changed by the new SEW/LMUL ratio. Use of the instruction with a new SEW/LMUL ratio that would 
result in a change of VLMAX is reserved. Use of the instruction is also reserved if vill was 1 beforehand. 
Implementations may set vill in either case.
```
Based on the above explanation, for the vsetvl or vsetvli instructions when rs1 = x0 and rd = x0:
(1) If the new SEW/LMUL ratio differs from the SEW/LMUL ratio in the vtype register;
(2) If the vill bit in the vtype register is already set to 1;
In either of these cases, if an implementation chooses to set the vill bit to 1, must it also clear all other bits in the vtype register and the vl register to 0?

gfavor, 2025-04-27:
The Unpriv manual states "The behavior upon decoding a reserved instruction is UNSPECIFIED."  So, in short, the behavior is UNSPECIFIED for both 1 and 2.

aswaterman, 2025-04-28:
Yes.  The spec says that "When the `vill` bit is set, the other XLEN-1 bits in `vtype` shall be
zero", which holds regardless as to why `vill` was set.  And of course `vl` can only hold legal values, and any positive value is illegal under an invalid configuration.

closed 2025-04-30 by shengqilong