riscv/riscv-isa-manual #1990: Which exception needs to be raised for accessing supervisor CSR which does not have matching VS CSR when V=1.

harsha-gunturi, 2025-04-27:
Hi 

As per specification,
Some standard supervisor CSRs (senvcfg, scounteren, and scontext, others) have no
matching VS CSR. These supervisor CSRs continue to have their usual function and accessibility even
when V=1, except with VS-mode and VU-mode substituting for HS-mode and U-mode. Hypervisor
software is expected to manually swap the contents of these registers as needed.

1.  Accessing senvcfg, scounteren or scontext in VS mode is permitted as HS substitutes VS and Hypervisor Software should take care of this register to manually swap.
2. Accessing senvcfg, scounteren or scontext CSR in VU mode, should raise Illegal Instruction or Virtual Instruction Exception? As U substitutes VU, accessing these CSRs in U mode will result in Illegal Instruction Exception so accessing the same CSRs in VU mode should also raise Illegal Instruction Exception.

Could you clarify the above statements?

radimkrcmar, 2025-04-27:
1. Yes. 
2. Virtual-instruction exception.
```
When V=1, a virtual-instruction exception (code 22) is normally raised instead of an illegal-instruction
exception if the attempted instruction is HS-qualified but is prevented from executing when V=1 either
due to insufficient privilege [...]
An instruction is HS-qualified if it would be valid to execute in HS-mode
```
senvcfg/scounteren/scontext CSR accesses are HS-qualified and VU doesn't have sufficient privilege.

harsha-gunturi, 2025-04-28:
@radimkrcmar, what does

harsha-gunturi, 2025-04-28:
Hi @gfavor and @radimkrcmar, 
Thanks for the quick response and clarification.

closed 2025-04-28 by harsha-gunturi