riscv/riscv-isa-manual #1992: Svvptc - how to implementation?

rock-ifly, 2025-04-28:
> Depending on the microarchitecture, some possible ways to facilitate implementation
of Svvptc include: not having any address-translation caches, not storing Invalid PTEs
in the address-translation caches, automatically evicting Invalid PTEs using a
bounded timer, or making address-translation caches coherent with store
instructions that modify PTEs.

RISCV spec, Vol II - privileged Arch, ch17, Svvptc gives some possible ways to implement Svvptc extension:
a) not having any address-translation caches, 
b) not storing Invalid PTEs in the address-translation caches, 
c) automatically evicting Invalid PTEs using abounded timer, 
d) making address-translation caches coherent with store instructions that modify PTEs.

Q1: How to understand these proposals? Just one of them is OK enough, like just a or b or c or d, to implement the Svvptc extension? or all of them (a+b+c+d) is the correct way to implement the Svvptc extenstion?

Q2: why we need such an extension? what is the benifit? and how to get the synchronization between the cores in the system? Just a store instruction cann't do that.

rock-ifly, 2025-04-29:
> 1. The mechanism used by the implementation to achieve the property is specific to the implementation. This note provides some suggestions. It is usually not necessary to implement all of them . Each microarchitecture needs to reason about its implementation and choose the best strategy.
> 2. The benefits of the extension are outlined here -https://lists.riscv.org/g/tech-privileged/message/1643

OK, it just needs to choose a proper strategy, not all of them.

Thanks a lot.

closed 2025-04-29 by ved-rivos

rock-ifly, 2025-04-29:
one more question, what does the "invalid pte" mean?
A: pte.valid=0, means these pte are not valid

rock-ifly, 2025-04-29:
seems an proper implemntation of Svvptc
1) not cache a pte.v=0 entry in the address-translation cache
    --1a) done by making pte.v=0 unmatch, and replace the pte with v=0 as soon as possible in the address-translation cache
    --1b) how to make pte.v=0 for the pte in the ATC? should it be the sfence.vma instruction? **It seems that software issue an  sfence.vma if necessay**
2) store instruction change the pte, like pte.v = 0 -->1, without issuing a sfence.vma
    --2a) seems done by software, but if the software want to use the newly modified PTE, it may still get the old, stale PTE already in TLB
    -- 2b) **it seems that software needn't issue a sfence.vma for pte modify instruction, like store for a pte date, it depends for the Svvptc.**
3)  who make sure the correction? Software? MMU cann't do anything if software goes wrong. like the 2a case.

ved-rivos, 2025-04-29:
1. The goal of the Svvptc extension is to obviate the need for an sfence.vma when marking an invalid PTE valid (changing V from 0 to 1). This is provided that an sfence.vma was used previously when the PTE was made invalid (changing V from 1 to 0).
2. The Svvptc extension only obviates need for an sfence.vma when making a PTE valid. If the permission attributes (rwx) are changed then a sfence.vma is required. 
3. The implementation and software need to follow the rules outlined in the Svvptc extension for correctness.

jillleon007, 2025-05-06:
> 1. Svvptc extension obviates the need for an sfence.vma when making a PTE valid (changing V from 0 to 1). Making a PTE invalid (change V from 1 to 0) continues to require a sfence.vma.
> 2. If the permission attributes (rwx) in a PTE are changed then a sfence.vma is required.
> 3. The implementation and software need to follow the rules outlined in the Svvptc extension for correctness.

In the Svvptc extension，it said that：Depending on the microarchitecture, some possible ways to facilitate implementation of Svvptc include: 
1.not having any address-translation caches, not storing Invalid PTEs in the address-translation caches,
2. automatically evicting Invalid PTEs using a bounded timer, 
3. or making address-translation caches coherent with store instructions that modify PTEs.

I think the option 1 is implemented without TLB，but without the TLB，the performance would be lower I think.
Option 2,the bounded timer is a unreachable option for the hardware, designer can't know what is the correct bounded time.  
Option 3, does this mean that we need some new instructions to modify TLB directly？

ved-rivos, 2025-05-06:
I think in your numbered list you include two separate options as part of 1. If an implementation does not have a TLB or ATC then the issue is moot. For an implementation with a TLB/ATC, there is the option of "not storing invalid PTEs in the address-translation caches". This is likely the most common implementation. Option 3 implies an ATC/TLB design where the TLB/ATC entries are tracked by the coherence protocol such that a modification to the PTE in memory would cause an invalidating snoop/probe to the cached PTE in a TLB/ATC.

rock-ifly, 2025-05-21:
OK, let's take the common way to implement, "not storing invalid PTEs in the address-translation caches".

Thanks a lot. @ved-rivos