riscv/riscv-isa-manual #1995: vslideup tail

jiahzhang, 2025-04-30:
Since the "start element" is max(shift amount, vstart), should there still be a tail when `shift amount > vl > vstart`? There is effectively no body in this scenario. The spec says
```
vl <= i < VLMAX                Follow tail policy
```

aswaterman, 2025-04-30:
Yes, there is still a tail if the shift amount is greater than VL but VL < VLMAX.  (Of course, since undisturbed is a valid implementation of agnostic, it is legal for an implementation to pretend there is no tail in this case.)

jiahzhang, 2025-04-30:
Would it make more sense for the implementation to leave it undisturbed? I guess software shouldn't be doing this anyway

aswaterman, 2025-04-30:
The implementation should do whatever is most convenient for the implementation.  I don't think there's a single optimal answer here.

closed 2025-04-30 by jiahzhang