riscv/riscv-isa-manual #1998: Behaviour of out-of-range CLINT write

bantierr, 2025-05-02:
Hi,

I was wondering what the expected behavior of an out-of-range CLINT write is. 
When I say out-of-range, I mean that we are writing to to a, e.g. MSIP register, of an address that is in the CLINT memory range, but there is no hart in the system that maps to that MSIP.

So, for example, in a 2 core system, what should happen if I write to the MSIP of hart 3.

Should an exception be raised ? Will it have any side effect ? 

Spike does not raise an exception, an it has no effect on the system.

Thank you for you time and help,
Best,
Quentin

stnolting, 2025-05-02:
Raising a bus access error exception seems to be a good idea. But I think this is entirely platform-defined.

aswaterman, 2025-05-02:
Yeah, in general, the hart won't have enough information to generate synchronous exceptions in cases like this, so an interrupt is more likely.  But it is up to the platform.  (IIRC, the SiFive CLINT ignores writes to unimplemented addresses and returns 0 on reads.)

bantierr, 2025-05-03:
I got it, thanks a lot for all the answers !
Have a nice weekend :)

closed 2025-05-03 by bantierr