riscv/riscv-isa-manual #1999: Can hardware set the A bit of a PTE if the translation fails in step 5 or 6 of the address translation process?

jrahmeh, 2025-05-02:
Section 11.3.2 of the privilege says to "stop and raise a page fault" in steps 5 and 6 of the address translation process.
The "stop" indicates that we do not make it to step 7 where the A bit is updated.

However, the A bit may be set by a speculative access.  So why restrict the hardware from updating the A bit in the case of a page fault at steps 5 or 6?

Consider the following scenario where SVADU is implemented:

1. A leaf PTE has A=0, R=0, X=1 (not accessed, not readable, executable).
2. A speculative fetch (instruction prefetcher) does a translation, finds the PTE, translation is successful, A bit is set to 1.
3. A non-speculative load does a translation, finds the same PTE, takes a page fault at step 5. Does not get to step 7.

The page fault handler will find A==1 even though the hardware did not set it at item 3 above.  The software must deal with this case.  So why not allow the hardware to set the A bit in the case of a page fault in steps 5 and 6 of the address translation process?

Instead of "stop and raise a page fault" at steps 5 and 6, can we implement "stop and raise a page fault optionally updating the A bit"?

ved-rivos, 2025-05-02:
While for most implementations the page walk would get cancelled if the page walk detects a fault. The op that detected the fault may not may not cause a trap depending on whether it commits. 

However an implementation is free to set say A bit on a read-only, read-write, read-write-execute, read-execute, execute-only, or shadow-stack PTE as the nature of speculation is micro-architectural and the implementation can claim it was an unconfirmed speculative load or a speculative fetch or a speculative `sspopchk` that caused the A bit to be set. This does not need any changes to the process as stated in the Priv. manual as the specification already allows setting the A bit as a result of speculation.

aswaterman, 2025-05-02:
> This does not need any changes to the process as stated in the Priv. manual as the specification already allows setting the A bit as a result of speculation.

This is the key.  And a speculative access that causes the A bit to be set corresponds to a separate execution through this state machine.

jrahmeh, 2025-05-03:
Thank you for the clarification.  Closing.