riscv/riscv-isa-manual #2000: `addi` is supposed to be "constant time", including the immediate value, does this mean `mv` can't be optimized?

programmerjake, 2025-05-04:
as described below, `addi` is in the short list of instructions that aren't supposed to have execution latency depend on immediate values (to avoid leaking `pc` values), this means the cpu can't do anything special when the immediate is zero, which is a `mv` pseudo-instruction. register renaming cpus very commonly treat move instructions specially, by handling them in the rename stage so they effectively take 0 cycles to execute. this contradicts the requirement that `addi` can't have different latencies for different immediate values:
https://github.com/riscv/riscv-isa-manual/blob/242d17c4f62c4726e45b915df7d1332106e5a712/src/scalar-crypto.adoc?plain=1#L3954-L3959

I think there should be text added that an `addi` with a zero immediate is allowed to have a different latency than one with a non-zero immediate, but that its latency must still not depend on the values in the registers or on which particular non-zero immediate it has if it has one.

ved-rivos, 2025-05-04:
An addi with immediate value of zero implemented as a zero cycle move will still be a constant time instruction if it always takes 0 cycles for that instruction. If the implementation however had differing latency for the same instruction depending on data value then it would not qualify as constant time.

programmerjake, 2025-05-04:
the issue isn't that `mv` isn't constant-time, but that `addi`'s latency would depend on the immediate, which is not allowed according to the quoted text.

ved-rivos, 2025-05-04:
`addi` is not one instruction - each encoding of the immediate is a different instruction and each of them need to be constant time.

programmerjake, 2025-05-04:
> `addi` is not one instruction - each encoding of the immediate is a different instruction and each of them need to be constant time.

that renders meaningless the last part of the text I quoted from the isa manual above about not leaking the PC, so afaict that's not what the authors of that text implied.

programmerjake, 2025-05-04:
afaict that text was originally added in https://github.com/rvkrypto/riscv-zkt-list/commit/6207e14a4afe51706fa0ad06a15fbe7cb1b93e78 by @mjosaarinen

mjosaarinen, 2025-05-04:
In DIEL (daya-independent execution latency, i.e.. Zkt and Zvkt extensions) execution time can always depend on immediate values, register choices etc. But not *data*, i.e. contents of registers. Execution time can also depend on PC.

programmerjake, 2025-05-04:
so then that text should be deleted, or at least there should be a note explaining that, since it currently seems to imply that `addi` (and `lui` and `auipc`) must have the same latency for all immediate values.

mjosaarinen, 2025-05-04:
In DIEL (daya-independent execution latency, i.e.. Zkt and Zvkt extensions) execution time can always depend on immediate values, register choices etc. But not *data*, i.e. contents of registers. Execution time can also depend on PC (e.g. alignment.)

mjosaarinen, 2025-05-04:
This is a background section where we are trying to explain why you shouldn't put secrets into immediate values. How do you read _"instruction timing may leak information about immediates, ordering of input registers, etc."_ as them having to have the same latency? To me that seems exactly the opposite of what the text says. Immediates, ordering of input registers, etc can leak under DEAL.

mjosaarinen, 2025-05-04:
We are also very explicitly stating that: *"Also, DIEL does not apply to constant values specified in the instruction encoding such as the use of the zero register (x0), and, in the case of immediate forms of an instruction, the values in the immediate fields (i.e., imm, and uimm)."*

programmerjake, 2025-05-04:
> How do you read _"instruction timing may leak information about immediates, ordering of input registers, etc."_ as them having to have the same latency?

because I'm instead getting it from:
> There may be an exception to this 
 in systems where a binary loader modifies the executable for purposes of 
 relocation -- and it is desirable to keep the execution location (PC) secret.

which to me implies that since `addi` can be used in relocations and you're trying to keep the value of the relocation secret, that `addi`'s immediate must also be kept secret.

mjosaarinen, 2025-05-04:
Hmm. Yes that very last sentence can be confusing. This is a cryptography extension so the binary loading stuff is a diversion, so indeed perhaps that should go. In secure / cryptographic programming we never even try to hide the PC.