riscv/riscv-isa-manual #2003: Publish new set of ISA documents.

wmat, 2025-05-05:
I'd like to open the discussion around publication of a new set of ISA documents.  The last set is dated 20240411, so it's been over a year now.  I believe that any non-ratified material  has been removed.  The new process to trigger publication is the ratification of new ISA specs, of which we have had 17 ratifications in the last year.

I'd like to choose a date for publication if possible?

gfavor, 2025-05-05:
There has been a target from a few months ago (by the ICs) of producing a new public release of the ISA manuals in time for the RISC-V European Summit.  (And, in particular, a clean release with only ratified content.  For which I believe @aswaterman has completed the work to enable that.)

@aswaterman Any comment on any reason to hold off publishing new manuals this week?

gfavor, 2025-05-07:
@wmat To be extra clear (on behalf of the Unpriv and Priv IC chairs), please go ahead and publish new ISA manuals at the end of this week.  The public RISC-V website should also be updated to point to this release as the current official ISA Manuals.

closed 2025-05-12 by wmat