riscv/riscv-isa-manual #2012: Inconsistency of rs1/r1s

bgrayson-mips, 2025-05-08:
Throughout the unpriv spec, there are 1796 instances of `rs1` and 988 of `rs2`.
However, there are just a handful of occurrences of `r1sc` and `r2sc`, and all of them are used in the `cm.mva01s` and `cm.mvsa01` descriptions.

Is there a reason to switch to this different naming scheme? If so, it is self-**in**consistent, in that https://github.com/riscv/riscv-isa-manual/blob/ed0aaa296203a68f7b1f4a231c3971cd3ed28973/src/zc.adoc?plain=1#L260 uses `rs1c` for `cm.mva01s`, but https://github.com/riscv/riscv-isa-manual/blob/ed0aaa296203a68f7b1f4a231c3971cd3ed28973/src/zc.adoc?plain=1#L2248 uses `r1s`.

I can create a PR to clean this up, but wanted to verify this `r*s` inversion was unintentional before doing so.

bgrayson-mips, 2025-05-08:
Similarly, some parts use `r1s'` and others use `r1sc`. Is that also an intentional delta?

aswaterman, 2025-05-11:
cc @tariqkurd-repo

DavidYu360, 2025-05-12:
For instruction `cm.mva01s rs1', rs2'`, the names `rs1'`& `rs2'` fit the purpose(Move two s0-s7 registers into a0-a1).
The arguments in `cm.mva01s r1s', r2s'` are not the general meaning of source, so here the spec uses `r1s'` instead of `rs1'`.

[image]

IMO, why not introduce new arguments:  `cm.mva01s rd1', rd2'`, it fits the purpose of instruction better(Move a0-a1 into two different s0-s7 registers).

[image]

DavidYu360, 2025-05-12:
> Similarly, some parts use `r1s'` and others use `r1sc`. Is that also an intentional delta?

The names of operands end with `c` are all in sail pseudo code, maybe the `r1s'` is not a valid sail identifier, just use `r1sc` instead?

tariqkurd-repo, 2025-05-15:
I had a look into when that change was made - looks like it was from the public review feedback.

https://github.com/riscvarchive/riscv-code-size-reduction/commit/ace0ee0328f7de316a84d17b528e20b8fd28df3e

Before that they were `sreg1` and `sreg2`.
I don't have a strong feeling about what they should be called but agree that `rd1'` and `rd2'` would be more logical.
The `rs1c` and `rs2c` is for SAIL, but the `'` is used in the documentation to be consistent with `rs1'/rs2'/rd'`.

nazar-pc, 2026-05-09:
If it is arbitrary, why not fixing it?

Seems like a trivial change to me and I see no benefit in using `r1s'` instead of `rs1'`, especially if both are used in mnemonic of the same exact instruction. It is just confusing with zero benefit.

At very least use it in mnemonics one way or the other consistently.