riscv/riscv-isa-manual #2017: How to deal with ASID/VMID=0 in a system that cache Bare mode translation in TLB

supermessilzb, 2025-05-14:
Hi, 
In previleged spec, it says: 
> To select MODE=Bare, software must write zero to the remaining fields of satp (bits 30–0 when SXLEN=32, or bits 59–0 when SXLEN=64).

> When MODE=Bare, guest physical addresses are equal to supervisor physical addresses, and there is no
> further memory protection for a guest virtual machine beyond the physical memory protection scheme
> described in Section 3.7. In this case, the remaining fields in hgatp must be set to zeros.

In a system that cache bare-mode translation in TLB,  the ASID/VMID 0 may be used when MODE=Bare and when MODE != Bare.

My questions are:
1. What does hardware need to do for ASID 0, if a hart Switches to/from Bare mode?  For example, caching or sfence'ing?
2. The same to VMID.

Thanks！

aswaterman, 2025-05-15:
ASID 0 and VMID 0 are not special.  They are identical to all other VMIDs.

ASIDs and VMIDs are not active (and therefore not relevant) when MODE=Bare.  The fact that the ASID or VMID field happens to contain the value 0 is unimportant.

supermessilzb, 2025-05-15:
Got it. Thanks,  Andrew！

aswaterman, 2025-05-15:
You're welcome!

closed 2025-05-15 by aswaterman