riscv/riscv-isa-manual #2019: Confusing content of bitmanip extensions

pfusik, 2025-05-15:
This paragraph confuses me:

https://github.com/riscv/riscv-isa-manual/blob/584390d9d872835b1bc45ab9ea946b56db3ad485/src/b-st-ext.adoc?plain=1#L34

- Why does it mention "Public Review" if these extensions are long ratified?
- It gives an impression that Zbc is part of B. According to https://github.com/riscv/riscv-isa-manual/blob/584390d9d872835b1bc45ab9ea946b56db3ad485/src/b-st-ext.adoc?plain=1#L4 it is not.

Then comes a table. Why does it include Zbc, but not Zbkb, Zbkc, Zbkx ?

Last thing: Zbkb, Zbkc, Zbkx instructions are also specified in [scalar-crypto.adoc](https://github.com/riscv/riscv-isa-manual/blob/584390d9d872835b1bc45ab9ea946b56db3ad485/src/scalar-crypto.adoc). Is this intended? There are differences, for instance:
- `xnor` is Zbb 0.93 in `b-st-ext.adoc` but Zbb 1.0.0 in `scalar-crypto.adoc`
- `xperm` shows 7 low opcode bits in `b-st-ext.adoc` but 2+5 in `scalar-crypto.adoc`

gfavor, 2025-05-17:
The mention of "Public Review" is incorrect and needs to be cleaned up.

This chapter appears to be poorly titled since it describes not just what comprises the "B" extension, but describes all the "bit manip" extensions.  Or all the scalar crypto related "bit manip" extensions need to be separated out into their own chapter.

The "B" extension - corresponding to what misa.B represents - comprises just the three bit manip extensions specified.  Which all the more makes the title of the chapter a bad title (even if the scalar crypto "bit manip" extensions are separated out).

@aswaterman What do you think should be done as far as retitling and/or restructuring this chapter?

aswaterman, 2025-05-17:
This is a specific instance of a more general pattern, in which we've shoehorned a spec into the ISA manual without actually going through the effort of properly integrating it.

The authors of the various extensions are responsible for the integration.