riscv/riscv-isa-manual #2026: Dose execute Privilege Levels affect the function of sfence.vma/hfence.gvma.hfence.vvma?

AlexGJL, 2025-05-20:
1. if we use sfence in vs mode, dose this insrtuction invalid the tlb entry which match vs/hs mode, or just ignore the Privilege Levels and invalid all privilege level TLB entry?  how about execute this instruction in M mode?  how about execute hfence.gvma/hfence.vvma in M mode and HS mode?
2. dose mstatus.mpp and mstatus.mprv have effect on sfence.vma,hfence.gvma,and hfence.vvma,?

aswaterman, 2025-05-21:
As the hypervisor spec mentions, SFENCE.VMA's behavior is different when V=1 (in VS-mode) versus when when V=0 (in S-mode or M-mode).  When V=1, it synchronizes only updates to the second-stage (VS-stage) mappings and has no effect on the single-stage or G-stage mappings.  When V=0, it synchronizes only updates to the single-stage (S-mode) mappings and has no effect on the VS-stage or G-state mappings.

Executing SFENCE.VMA in M-mode has the same effect as executing SFENCE.VMA in S-mode.

MPRV only affects explicit memory accesses; SFENCE.VMA is not one of those.

AlexGJL, 2025-05-21:
thanks for your reply. if hardware cache va->pa translation mapping in TLB under M mode when mmu  is off, are there some methods or instructions to flush these TLB entries?

aswaterman, 2025-05-21:
This is a uarch question, not an ISA question. It is your job to figure this one out :)

closed 2025-05-21 by aswaterman