riscv/riscv-isa-manual #2027: CSR addresses missing

tzwaenn, 2025-05-20:
It seems that on adding some of the latest extensions, CSR addresses got edited away - at least I can't find the addresses of srmcfg or mctrctl and friends in either the extension specific section or the overview CSR map.

topperc, 2025-05-21:
srmcfg is in the overrivew file but commented out. https://github.com/riscv/riscv-isa-manual/blob/b3da54961e12024158132561a2e42c5b78bbc7ae/src/priv-csrs.adoc?plain=1#L346

ved-rivos, 2025-05-21:
It had been commented out with intent of uncommenting after Priv 1.13 release. It can be uncommented now.

closed 2025-05-21 by aswaterman

tzwaenn, 2025-05-22:
Seems the Smctr ones are still missing...

topperc, 2025-05-22:
> Seems the Smctr ones are still missing...

A separate issue was been filed for that and assigned to the correct person to fix it
https://github.com/riscv/riscv-isa-manual/issues/2033

tzwaenn, 2025-05-23:
Great, thanks!

PR #2035 (2025-05-21): Add srmcfg to CSR table
Resolves #2027