riscv/riscv-isa-manual #2029: What to do with "strided"?

kersten1, 2025-05-20:
The doc-sig has 2 remaining words that require a judge's ruling so I'm opening an issue in the main repo to get more exposure. 

From the discussion in the vale repo for [strided](https://github.com/riscv-admin/riscv-vale/issues/12).

Stride is an array thing, right?  So unit-stride, constant-stride - all those make sense to me.  But I'm not sure what just "strided" means in the following context?  I'm assuming it means constant-stride but I do not like to assume.... it just never works out well for me. 

[vector.adoc](https://github.com/riscv/riscv-isa-manual/blob/e5078e55ea977ef197066dcd3f7c8e09032cb348/src/v-st-ext.adoc?plain=1#L1377)

a little further down in line 1384, it says "Vector unit-stride operations" but in the next paragraph on line 1387, it says "Vector constant-strided operations". Is this correct? Shouldn't they both be either _stride_ or _strided_? Please advise....

@aswaterman responded:

Agreed that "unit-stride" and "constant-stride" are fine as-is.

The one instance of "constant-strided operations" can just be changed to "constant-stride operations", consistent with our use of the "constant-stride" compound adjective in a few other cases.

AFAICS, all remaining uses of "strided" are shorthand for "constant-stride". It makes sense to me to standardize on one or the other, with "constant-stride" being more verbose but perhaps more self-evident.

Since @kasanovic is more familiar with historical vector ISA naming conventions (and authored most of this text), I'd like him to sign off.

kasanovic, 2025-05-20:
I think "constant-stride"  is fine as the standard term.    I do have one lingering concern that there's no clear way of referring to "non-unit-but-constant" stride, as one reading is that "constant-stride" include unit-stride as a special case.  The vector machine history is muddled here as some machines did not have separate unit-stride encodings.  I think it is actually OK for RISC-V ISA convention to be to use "Constant-stride" to refer to only the strided instruction encodings (which do allow the stride value passed in a register to be unit stride).

aswaterman, 2025-05-21:
SGTM.  @kersten1 if you prepare a PR, I'd be happy to review it.

closed 2025-05-22 by aswaterman