riscv/riscv-isa-manual #2039: Some questions about PTE.D Hardware Behavior

haosun86, 2025-05-22:
Hi,

I have listed some scenarios, could you please help confirm whether hardware behavior is right?
(1) 
**Program:**
store -> cause PTE.D update
load -> read same physical address of that PTE (above)

**Hardware Benavior**
Load must observe PTE.D update

(2) 
**Program:**
store -> cause PTE.D update
load -> same physical page with store, and TLB miss with table walk

**Hardware Benavior**
Load's table walk may not observer the PTE.D update

(3) 
**Program**
load0 -> Speculative VS level PTE.A update, that cause G-Stage PTE.D update
load1 -> read same physical address of that G-Stage PTE (above)

**Hardware Behavior**
Load1 must observe G stage PTE.D update

(4) 
**Program**
load0 -> Speculative VS level PTE.A update, that cause G-Stage PTE.D update

**Hardware Behavior**
If load0 is flushed due to branch mis-prediction, G-Stage PTE.D update can still perform.

Looking forwarding to your reply
Thanks

aswaterman, 2025-05-22:
Agreed with (1).

I don't understand the significance of (2), that the load's table walk might not observe the PTE.D bit, because the PTE.D bit has no effect on loads.

Agreed with (3).

For (4), yes, but I would be more precise.  You wrote "PTE.D update _can_ still perform".  I'd say "_must_ be performed _if_ VS-stage PTE.A is set."  But yes, your broader point is correct: it's valid for misspeculation to cause the VS-stage PTE.A to be set, in turn causing the G-stage PTE.D to be set.

Agreed with (5).

haosun86, 2025-05-23:
Thanks so much.

For the scenario (5), I think I have some typos, and I have corrected them, which is shown below.
**Program**
store -> cause PTE.D update, but it is flushed due to interrupt

**Interrupt_handler:**
load -> read same physical address of that PTE (above)

**Hardware Behavior**
Load must observe PTE.D update.

**But I have one question:**
**Could you please help explain why the Load needs to observe the PTE.D update? After all, that store is flushed, and there is no program order relationship between that store and the load in the interrupt_handler.**

aswaterman, 2025-05-23:
> Could you please help explain why the Load needs to observe the PTE.D update?

My argument is that this case still follows from the requirement that the PTE update must be observed in program order by the local hart.  Even though the store did not retire, the store is the reason the PTE update occurred, and so the PTE update itself is in program order before the load.

This case should not be materially more difficult to implement than the other cases (and in fact it should be easier in many implementations, since the interrupt itself is at least partially serializing).

haosun86, 2025-05-23:
Thanks so much.  It is clear!

Could you please help check the final scenario:
**Proragm**
store -> PTE.W is zero, and store does not have write permission. And PTE.D is clear. 

**Hardware Behavior**
Will the PTE.D be updated or not?

haosun86, 2025-05-24:
Thanks so much.

closed 2025-07-04 by ved-rivos