riscv/riscv-isa-manual #2040: VSTIMECMP register access when both mcountern.TM and hcountern.TM

harsha-gunturi, 2025-05-22:
As per spec,

In addition, when the TM bit in the hcounteren register is clear, attempts to access the vstimecmp register (via stimecmp) while executing in VS-mode will cause a virtual-instruction exception if the same bit in mcounteren is set. "When this bit and the same bit in mcounteren are both set, access to the vstimecmp register (if implemented) is permitted in VS-mode."
The statement in quotes above, this means "access to vstimecmp(via stimecmp)" is permitted or we can access vstimecmp in VS mode when both  mcountern.TM and hcountern.TM are set? But VS CSRs can't be accessed directly in VS mode. 

Could you please clarify this?

radimkrcmar, 2025-05-22:
`via stimecmp` implicitly implies to the second sentence as well -- it's the classic mix-up between CSR (interface) and CSR (state).

vstimecmp CSR access in VS mode is not affected by the TM bit.

harsha-gunturi, 2025-05-22:
@radimkrcmar , Thanks for the confirmation.