riscv/riscv-isa-manual #2041: Two questions about vector unit-stride mask load/store instruction

wusong-ai, 2025-05-22:
1、For 'vlm.v', must 'vd' be the V0 register? Similarly, for 'vsm.v', is 'vs3' required to be V0?
2、Is it required to generate an illegal instruction exception if nf[2:0] is not zero in 'vlm.v' and 'vsm.v'?

aswaterman, 2025-05-22:
1. No, these instructions may target any vector register.
2. These instructions are reserved when `nf` is nonzero.  So, generating an illegal-instruction exception is the recommended behavior, but is not strictly speaking required.

wusong-ai, 2025-05-23:
Thanks，and I have another question. that for the vlm.v and vsm.v instructions, should emul (where emul = sew × lmul / eew) still be treated as 1 during execution, even if its computed value differs from 1?

wusong-ai, 2025-05-23:
Thanks, and I have another question. 
For the 'vlm.v' and 'vsm.v' instructions, should emul (where emul = sew × lmul / eew) still be treated as 1 during execution, even if its computed value differs from 1?

aswaterman, 2025-05-23:
Yes, the spec explicitly says that these instructions' data operands are EMUL=1.