riscv/riscv-isa-manual #2047: Restrictions on "the code to retry a failing LR/SC sequence"

newpavlov, 2025-05-27:
Currently the spec in the ["Eventual Success of Store-Conditional Instructions"](https://github.com/riscv/riscv-isa-manual/blob/main/src/a-st-ext.adoc#eventual-success-of-store-conditional-instructions) section states the following:

>The code to retry a failing LR/SC sequence can contain backwards jumps and/or branches to repeat the LR/SC sequence, but otherwise has the same constraint as the code between the LR and SC.

If we can not satisfy this condition, then the retry loop is considered "unconstrained" and it "might never succeed on (some) implementations". So does it mean that the following code "might never succeed"?
```
foo:
        lw      a1, 0(a0)
.LBB0_1:
        # Code which processes `a1` without modifying it
        # and returns result in `a2`.
        # Can contain more than 16 instructions and
        # can use non-I instructions
        ...

        mv      a3, a1
        lr.w.aq a1, (a0)
        bne     a1, a3, .LBB0_1
        sc.w.rl a3, a2, (a0)
        bnez    a3, .LBB0_4
        ret
```

This means that compilers have to be conservative and insert retry loops for ["weak" compare exchanges](https://doc.rust-lang.org/std/sync/atomic/struct.AtomicU32.html#method.compare_exchange_weak) (which were specifically introduced to efficiently support targets with LL/SC instructions). It results in a suboptimal codegen, making the "weak" and "non-weak" variants effectively the same. For example, see this LLVM issue: https://github.com/llvm/llvm-project/issues/141488

In my opinion, the restriction is too strict in its current form. Why code outside of the the LR/SC sequence should influence its forward progress guarantees?

aswaterman, 2025-05-27:
That loop is unconstrained, and it isn't guaranteed to make progress without fallback code.  This is why GCC's RISC-V backend emits the same code for `atomic_compare_exchange_weak` as it does for `atomic_compare_exchange_strong`, much like the x86 backend.  (I wasn't familiar with what LLVM's RISC-V backend is doing, but it appears to have taken the same tack.)

> For example, see this LLVM issue: https://github.com/llvm/llvm-project/issues/141488

I agree with what @asb wrote in https://github.com/llvm/llvm-project/issues/141488#issuecomment-2913515105

> Why code outside of the the LR/SC sequence should influence its forward progress guarantees?

At risk of stating the obvious, it's because implementations are allowed to rely on the definition of constrained LR/SC loops.  For example, they might only be able to make progress in that code between the LR and SC (inclusive) is resident in the I$ at the same time.  Unbounded code could cause capacity or conflict misses that evict the I$ line containing the BNE or SC, voiding the eventuality guarantee on these machines.

> In my opinion, the restriction is too strict in its current form.

For better or worse (and I'd argue "for better"), this definition is what we've got to work with.

newpavlov, 2025-05-27:
It's really disappointing (to put it mildly) that "weak" and "strong" are the same on RISC-V. From the programmer's point of view it effectively erases a big part of advantage of having LL/SC instructions in the ISA. And no, suggesting for people to write assembly manually is not a solution to this.

>For example, some implementations might only be capable of making progress if the code between the LR and SC (inclusive) is resident in the I$ at the same time. Unbounded code after .LBB0_1 could cause capacity or conflict misses that evict the I$ line containing the BNE or SC, voiding the eventuality guarantee on these machines.

How is it different from executing an LR/SC sequence after a big chunk of linear code? In both situations the sequence is not in I cache and hardware should (in my opinion) handle it appropriately. In other words, what is a difference between executing LR/SC sequence without retrying (and maybe calling the function again some time later) and doing it in a big retry loop? Are you saying that on some hardware implementations the sequence may always fail in the former case (e.g. if we go unlucky and the sequence resides in different cache lines)? This looks like yet another case of granting flexibility to hardware vendors at the cost of compiler authors and programmers.

Feel free to close this issue if you think that this is fine and no changes are needed.