riscv/riscv-isa-manual #2070: Accessing Unimplemented Indirect CSR from VS mode via VSIREG

harsha-gunturi, 2025-06-08:
1. vsiselect from M mode is programmed to contain an unimplemented iCSR. (As it an unimplemented iCSR access should result in an exception)
2. Accessing vsireg from VS mode 

Step2, As we are accessing VS CSR(vsireg) from VS mode, virtual exception should be raised? or As the access is not HS qualified (access to vsireg in HS mode when vsiselect is programmed to contain an unimplemented iCSR) Illegal insruction exception should be raised?

Could you please clairfy on whether which exception should be raised.

radimkrcmar, 2025-06-09:
If the vsiselect was set by M-mode, HS-mode, or VS-mode doesn't matter.  The set of legal values is the same.
The behavior is UNSPECIFIED:
```
The behavior upon accessing vsireg* from M-mode or HS-mode, or accessing sireg* (really vsireg*)
from VS-mode, while vsiselect holds a value that is not implemented at HS level, is UNSPECIFIED.
```
It's recommended to raise an illegal-instruction exception, but it's software's responsibility to know the implementation.

harsha-gunturi, 2025-06-09:
@radimkrcmar,

I am trying to access "vsireg" in VS mode when vsiselect holds a unimplemented VS indirect CSR.
As I am accessing VS CSR - vsireg in VS mode, shouldn't this result in a virtual exception trap ?
From spec,
"When V=1, an attempt to read or write a VS CSR directly by its own separate CSR address causes a virtual-instruction exception"

gfavor, 2025-06-10:
The spec excerpt in https://github.com/riscv/riscv-isa-manual/issues/2070#issuecomment-2956354584 is correct.  This is different than the cases covered in https://github.com/riscv/riscv-isa-manual/issues/2070#issuecomment-2955190829 because accessing a VS CSR address in VS/VU modes is different than accessing that CSR via its S/HS CSR address, i.e. two different CSR addresses for the same underlying physical CSR register.

In general, VS CSR addresses are for use by a hypervisor and should not be used by a guest - unless we're in a nested hypervisor situation, in which case the guest hypervisor will try to use the VS CSR address and that will trap to the host hypervisor.

Unless there is a further question, please close this issue.