riscv/riscv-isa-manual #2079: Implement of "Smrnmi extension"

wusong-ai, 2025-06-12:
If only the "Smrnmi extension" is implemented, is it necessary to follow the implementation scheme described in "The RISC-V Instruction Set Manual: Volume II Privileged Architecture 3.1.6.2. Double Trap Control in mstatus Register" (i.e., updating mn* CSRs and using the RNMI handler to handle exceptions) when the processor is in M mode and mnstatus.NMIE = 1?
 
[image]

When implementing as upon, in the scenario where the processor is in M mode with mnstatus.NMIE = 0 and an exception occurs, the descriptions in Volume II's "3.1.6.2 Double Trap Control in mstatus Register" and "8.5. RNMI Operation" appear to conflict. For microarchitecture implementation, should this case be handled by halting the hart or by updating CSRs and processing the exception via the RNMI handler？

ved-rivos, 2025-06-12:
> If only the "Smrnmi extension" is implemented, is it necessary to follow the implementation scheme described in "The RISC-V Instruction Set Manual: Volume II Privileged Architecture 3.1.6.2. Double Trap Control in mstatus Register"

It is only when Smrnmi and Smdbltrp are both implemented that the rules highlighted in your post apply.

ved-rivos, 2025-07-04:
Hope that helped. Please ask if there are further questions.