riscv/riscv-isa-manual #2086: Compressed shift hints are weird

Timmmm, 2025-06-17:
There are a few weird things with the compressed shift instruction wrt hints.

1. On RV64 `shamt[5]=1` aren't "reserved"

> For RV32C, shamt[5] must be zero; the code points with shamt[5]=1 are designated for custom extensions. For RV32C and RV64C, the shift amount must be non-zero; the code points with shamt=0 are HINTs.

`are designated for custom extensions` should just be `are reserved` surely?

2. The [hints table](https://riscv-specs.timhutt.co.uk/spec/20240411/unpriv-isa-asciidoc.html#rvc-t-hints) is confusing...

* `RV32 and RV64 only` is redundant - presumably it was missed when removing RV128.
* `C.SLLI64` etc aren't real instructions? Is that some remnant from RV128 too?

3. Only left shifts to x0 are hints?

`C.SLLI` with `rd=x0` are hints, but apparently that isn't the case for `C.SRLI` or `C.SRAI` with `rd=x0`? That doesn't make any sense does it? Those instructions also have no effect so they could be hints too surely?

topperc, 2025-06-17:
> Only left shifts to x0 are hints?
> C.SLLI with rd=x0 are hints, but apparently that isn't the case for C.SRLI or C.SRAI with rd=x0? That doesn't make any sense does it? Those instructions also have no effect so they could be hints too surely?

C.SLLI has a 5-bit rd field. C.SRLI and C.SRAI have a 3-bit rd field that can only encode x8-x15.

gfavor, 2025-06-17:
Custom vs. Reserved is an important distinction in the achitecture.  Custom code points are permanently set aside for use to encode custom instructions and are guaranteed to never be used to encode future standard instructions.  Conversely, Reserved code points may be used to encode future standard instructions and should not be used for custom instructions (such an implementation would be termed a Non-Conforming implementation).

It would appear that "RV32 and RV64 only" are no longer appropriate to state.  Although, then again, the presence vs lack of this text is making a forward looking statement, i.e. that code points with this text are not guaranteed to remain as custom code points in a future RV128.  In other words, the presence of this text avoids committing a future RV128 to treat these as custom code points.

Timmmm, 2025-06-17:
>  C.SRLI and C.SRAI have a 3-bit rd field that can only encode x8-x15.

Ah of course, thanks!

> Custom code points are permanently set aside for use to encode custom instructions and are guaranteed to never be used to encode future standard instructions.

Ah ok that makes sense.

> the presence of this text avoids committing a future RV128 to treat these as custom code points.

I think RV128 would be a new base architecture so it can do that anyway.

Timmmm, 2025-06-19:
Fixed in #2088

closed 2025-06-19 by Timmmm