riscv/riscv-isa-manual #2087: access scountinhibit register from vs or vu mode

shengqilong, 2025-06-17:
```
When Supervisor Counter Delegation is enabled, attempts to access scountinhibit from VS-mode 
or VU-mode raise a virtual-instruction exception.
```
What does 'When Supervisor Counter Delegation is enabled' specifically refer to? Does it mean both menvcfg.CDE=1 and mcounteren[i]=1 being set?

gfavor, 2025-06-17:
Earlier text says "When the Smcdeleg/Ssccfg extension is enabled (menvcfg.CDE=1), ..."

An analogous earlier text says "An attempt from VS-mode to access any sireg* (really vsireg*) raises an illegal-instruction
exception if menvcfg.CDE = 0, or a virtual instruction exception if menvcfg.CDE = 1."  The paragraph of text in question here is expressing similar intent.

shengqilong, 2025-06-20:
```
Smcdeleg/Ssccfg defines a new scountinhibit register, a masked alias of mcountinhibit. For counters 
delegated to S-mode, the associated mcountinhibit bits can be accessed via scountinhibit. For counters 
not delegated to S-mode, the associated bits in scountinhibit are read-only zero.

When menvcfg.CDE=0, attempts to access scountinhibit raise an illegal instruction exception. 
When Supervisor Counter Delegation is enabled, attempts to access scountinhibit from VS-mode 
or VU-mode raise a virtual-instruction exception.
```
The above description is confusing. Here's my understanding - please confirm if it's correct.
（1）access scountinhibit from u mode raise illegal instruction exception
（2）access scountinhibit from vu mode raise virtual instruction exception
（3）when menvcfg.CDE=0, access scountinhibit from m/debug/hs/vs mode raise illegal instruction exception
（4）when menvcfg.CDE=1, access scountinhibit from vs/vu mode raise virtual instruction exception

Is scountinhibit register accessible in M/H/S/debug modes only when menvcfg.CDE=1?

In what cases does reading scountinhibit return 0?

gfavor, 2025-06-24:
Closing since it seems like this issue question is resolved.