riscv/riscv-isa-manual #2096: questions about exceptions in sscsrind/smcdeleg/sscfg extension

shengqilong, 2025-06-19:
```
The behavior upon accessing vsireg* from M-mode or HS-mode, or accessing sireg* (really vsireg*) from VS-mode,
while vsiselect holds a value that is not implemented at HS level, is UNSPECIFIED.

It is recommended that implementations raise an illegal-instruction exception for such accesses, 
to facilitate possible emulation (by M-mode) of these accesses.
```

When sscsrind/smcdeleg/sscfg extension is enabled, sireg3-6 have no indirectly mapped registers. Are the following exceptional cases reasonable?
（1）m/hs/debug mode access sireg3-6 raise illegal instruction
（2）vs mode access sireg3-6(really vsireg3-6) raise illegal instruction 
（3）vu mode access sireg3-6(really vsireg3-6) raise virtual instruction

gfavor, 2025-06-20:
The excerpt from the spec is not relevant to any of the questions asked.  But the earlier spec section about Supervisor-level CSRs
 IS relevant - which essentially recommends raising an illegal-instruction exception from M/HS modes.  And, since instruction accesses to these unimplemented registers are not HS-qualified, I believe U/VU-mode accesses should also raise an illegal-instruction exception.

shengqilong, 2025-06-20:
```
Furthermore, while vsiselect holds a value in the range 0x40-0x5F:

* An attempt to access any vsireg* from M or S mode raises an illegal instruction exception.
* An attempt from VS-mode to access any sireg* (really vsireg*) raises an illegal-instruction exception 
  if menvcfg.CDE = 0, or a virtual instruction exception if menvcfg.CDE = 1.

```
When accessing vsireg* registers, is it necessary to consider the permissions controlled by mcounteren and hcounteren?

gfavor, 2025-06-27:
Closing since this question appears to resolved.

closed 2025-06-27 by gfavor