riscv/riscv-isa-manual #2103: Is it *required* that no element is written when vstart >= vl?

Timmmm, 2025-06-26:
The Vector chapter seems a bit inconsistent about writing to elements with indices less than `vstart`, especially when `vstart >= vl`.

First, you have to start at `vstart`:

> All vector instructions are defined to begin execution with the element number given in the vstart CSR, leaving earlier elements in the destination vector undisturbed

But sometimes you don't:

> Except where noted above, vector instructions are allowed to overwrite their inputs, and so in most cases, the vector instruction restart must be from the vstart element index. However, there are a number of cases where this overwrite is prohibited **to enable execution of the vector instructions to be idempotent and hence restartable from an earlier index location.**

Then, there's a general condition that you don't do anything if `vstart >= vl`:

> When vstart ≥ vl, there are no body elements, and no elements are updated in any destination vector register group, including that no tail elements are updated with agnostic values.

For [whole register vector move](https://riscv-specs.timhutt.co.uk/spec/riscv-isa-release-7beccc2-2025-06-20/riscv-unprivileged.html#_whole_vector_register_move) instructions it's `evl` instead of `vl`, but does that mean if `vstart>=evl` then you *can't* do the vector move, but if `vstart=evl-1` then you can do the entire vector move ignoring the exact value of `vstart`?

That seems really weird. Surely the intent of this instruction is that it can be handled in the renamer without having to worry about `vstart`?

The spec could really do with some RFC-style `MUST`s in here to make it obvious if it is intentional, and probably some note explaining why. I don't really see why you shouldn't be allowed to do the whole vector register move irrespective of the `vstart` value.

aswaterman, 2025-06-27:
> The Vector chapter seems a bit inconsistent about writing to elements with indices less than vstart, especially when vstart >= vl.

I agree the rules are quirky, but where's the inconsistency?  The third excerpt you quoted gives the answer in the case of `vstart` >= `vl`; "no elements are updated in any destination vector register group".

> but does that mean if vstart>=evl then you can't do the vector move

Right, per the above.

> but if vstart=evl-1 then you can do the entire vector move ignoring the exact value of vstart

No, in this case elements [0, evl-2] are not written.

The "whole register" terminology refers to the fact that LMUL and `vl` do not affect these instruction's behavior; `vstart` still applies (because nothing in the spec says it doesn't apply :-))

Timmmm, 2025-06-27:
But you are *allowed* to ignore `vstart` in that case because

> However, there are a number of cases where this overwrite is prohibited to enable execution of the vector instructions to be idempotent and hence restartable from an earlier index location.

And for that instruction the source and destination cannot overlap. Is that right? 

Was it not the intention that this instruction would be handled in the register renamer (for designs that have one)?

radimkrcmar, 2025-06-27:
An implementation that handles the whole vector move in the renamer can only allow `vstart=0`.

The manual doesn't seem to allow implementations to ignore the `vstart`:
``` 
* The prestart elements are those whose element index is less than the initial value in the vstart
register. The prestart elements do not raise exceptions and do not update the destination vector
register.
```

(I would also like to know what your quote means... My current interpretation: on a trap, an implementation can choose to provide a `vstart` that is lower (e.g. zero), if re-executing the instruction with that `vstart` would yield the same result, even though some destination registers were already overwritten.)

Timmmm, 2025-06-27:
Ah yes you're right

> For example, some implementations will never take interrupts during execution of a vector arithmetic instruction, instead waiting until the instruction completes to take the interrupt. Such implementations are permitted to raise an illegal-instruction exception when attempting to execute a vector arithmetic instruction when vstart is nonzero. 

... but yeah these two things seem to be in direct contradiction:

> The prestart elements do not raise exceptions and do not update the destination vector register.

and

> to enable execution of the vector instructions to be idempotent and hence restartable from an earlier index location.