riscv/riscv-isa-manual #2107: About vstval update method

haosun86, 2025-06-28:
Hi,

In the RISC-V Specification, it does not have description about when to update vstval like mtval/stval.
I am not sure whether the following statements are right for vstval:

（1）When a trap is taken into VS-mode，vstval is either set to zero or written with exception-specific information.
Otherwise, vstval is never written by the implementation.
（2）The hardware platform will specify which exceptions must set vstval informatively, which may unconditionally set to to zero, 
and which may exhibit either behavior, depending on the underlying event that caused the exception.

Looking forwarding to your reply.
Thanks
Hao Sun

gfavor, 2025-06-28:
Since vstval substitutes for stval when V=1, exceptions that modify stval will instead modify vstval.  Similarly, instructions that normally read or modify stval will actually access vstval instead.

haosun86, 2025-06-29:
For example, if there is one virtual instruction trap, the trap will be taken to HS mode. Should vstval be updated?

gfavor, 2025-06-29:
If V=1 and the trap is being taken within the VM (i.e. to VS-mode), then vstval (which is substituting for stval while V=1) is modified.  But if the trap is being taken out of the VM to V=0 (i.e. to HS-mode), then the actual stval is modified.

allenjbaum, 2025-06-29:
Looking at it from. code perspective:
    Handlers of a trap taken into any Smode (HS, VS, S) should look at
STVAL.
        When a trap is taken in VS mode, VSTVAL will have been updated and
accessed using the STVAL address
        When a trap is taken in HS or S mode, STVAL is updated as usual.
Bottom line is when V=1, STVAL and VSTVAL are effectively swapped, so
raccesses of STVAL (explicit CSR ops or implicit trapping updates) actually
access VSTVAL.

gfavor, 2025-07-01:
Closing since I believe this issue has been answered.