riscv/riscv-isa-manual #2108: [RVV] Question about agnostic implementation

Steven-Li-Xiaogang, 2025-07-01:
> When a set is marked undisturbed, the corresponding set of destination elements in a vector register group retain the value they previously held.

> When a set is marked agnostic, the corresponding set of destination elements in any vector destination operand can either retain the value they previously held, or are overwritten with 1s.

As RVV spec mentioned, we have two choice to implement destination elements which are marked as agnostic. can we have some instructions with agnostic marked elements retain the value in 'vd' register and some others overwrites marked elements in 'vd' with 1s?

Steven-Li-Xiaogang, 2025-07-02:
@aswaterman @gfavor 
Grant this flexibility in hardware implementation, seems be a challenge to verification engineer. If one can not get an uniformed process, tail/marked elements in 'vd' vector algorithm instructions may required to be handled one by one in verification environment.
For a verification environment which is setup to support co-simulation with RISC-V function/TLM model, vector register files will be used as checkpoint between model and HW, this kind of implementation would also be a challenge.

aswaterman, 2025-07-02:
Your implementation is free to use a strict model to simplify verification.