riscv/riscv-isa-manual #2129: [RVV] How to explain "Mask element past vl" for Vector Mask-Register Logical Instructions

Steven-Li-Xiaogang, 2025-07-13:
[image]
As text highlighted in above snapshot, new term "Mask elements past vl" is involved with tail-agnostic support, what's the difference from the description in other sections "remaining elements past vl".

Steven-Li-Xiaogang, 2025-07-13:
@aswaterman I suggest to revise the description "Mask elements" as it indicates body elements with v0.mask[i]=1 or all body elements if vm=1, otherwise it will be considered that tail elements should also be with v0.mask[i]=1.

gfavor, 2025-07-15:
The highlighted text is specific to vector mask logical instructions and is referring to per-element mask bits (in the mask result produced by these instructions) that are past vl.  These bits are of course subject to tail handling and specifically to the tail-agnostic policy.  And note that vm=1 is reserved for these instructions.

Whereas other references in the arch spec, for vector instructions in general, to "remaining elements past vl" are referring to the same issue and tail handling idea, but with just the difference of referring to actual elements and the handling being subject to the tail policy selected by vtype.vta.

So I don't believe there is any actual ambiguity.  "elements past vl" and "remaining elements past vl" are saying the same thing.  The use of "mask elements" in the former case is simply a reflection of the fact that vector mask logical instructions produce a mask (and never have a mask source operand).

gfavor, 2025-07-21:
Closing since it seems like this issue has run its course and been resolved.

closed 2025-07-21 by gfavor