riscv/riscv-isa-manual #2133: Does sfence need to consider the pagesize? clear one entry, or all the tlb entries with the matching va & va_pagesize_mask entries?

rock-ifly, 2025-07-15:
Hi,everyone, I have a question: Should the implementation of sfence be aware of the pagesize? 

What I mean is, in page table walk, a 2M address page (non-leaf) is an invalid page. 
1）We store the fault information and lookup level together in the TLB entry. 
2）The OS extracts the fault address from stval and initiates a sfence by va operation. The VA width is [38:12]. 
3）We compare the flushed VA with the VA stored in TLB. 
3.1）If the VA [39:12] matches, we clear that TLB entry. 
Notice that, with this matching method, this clearing can only remove one 4K from the invalid 2M page, not all the faulty pages, neglecting considering the bigger pagesize (2MB/1GB) entry with the same VA[38:21] for 2MB pagesize, or  VA[38:30] for1GB pagesize.

Then, when sfence performs clearing, should the hardware match considering the pagesize in the TLB entry? 
For example,  a 2M page, if VA [39:21] matches, shoulde sfence clear that 2MB entry?

aswaterman, 2025-07-15:
For reference, the relevant statement in the spec is that SFENCE.VMA with rs1 != x0 "invalidates all address-translation cache entries that contain leaf page table entries corresponding to the virtual address in rs1".  i.e. it doesn't matter what size the leaf page is; it's still required to flush any TLB entry related to it.

rock-ifly, 2025-07-16:
> On the one hand, architecturally an address-specific sfence has no information about what page size or sizes may contain the address in question. And on the other hand, **any and all cached translations of whatever page size(s) that cover this address** must all be invalidated.

Does the "any and all cached translations of whatever page size(s) that cover this address" include walk cache entry? It seems better to flush the walk cache entries and the leaf page entries.

Thanks.

jillleon007, 2025-07-16:
> If the walk cache only contains only non-leaf entries, it need not be flushed when rs1 != x0.

But there is a gap, hardware only update scause/stval etc，How does OS know the VA is a leaf page or non-leaf page?