riscv/riscv-isa-manual #2134: How to handle non-idempotent regions accessed by vector unit-stride loads?

Qin0803, 2025-07-15:
In RV Privileged SPEC, it describes that
> While hardware should always be designed to avoid speculative or redundant accesses to memory regions marked as non-idempotent, it is also necessary to ensure software or compiler optimizations do not generate spurious accesses to non-idempotent memory regions.

> For non-idempotent regions, implicit reads and writes must not be performed early or speculatively, with the following exceptions. When a non-speculative implicit read is performed, an implementation is permitted to additionally read any of the bytes within a naturally aligned power-of-2 region containing the address of the non-speculative implicit read.

In specific implementations, multiple elements may be loaded at once for vector unit-stride loads. for example, vle8.v, it can load up to 16B of data in a single memory access. Could you please help me confirm whether the following scenario is allowed?

1)  If the accessed total size and address are not aligned, all elements are active, and the accessed regions is non-idempotent, it can be split into two parts for implementation. For example, if the address is 0x1fff and the total size is 16B, the first part loads one byte of data, and can the second part load 16 bytes of data (with an additional byte loaded)?

2) Similarly, for the vle8.v instruction, not all elements are active, and the accessed region is non-idempotent. Is it allowed to load all elements in a single memory access, regardless of whether the corresponding elements are active or inactive?

avakar, 2025-07-15:
You quote text referring to implicit reads, but then ask about vle8.v, which performs explicit access.

gfavor, 2025-07-15:
As avakar notes, the excerpted text is not relevant to the question.  The question is instead about explicit accesses to non-idempotent PMA regions - which in general for all loads (scalar and vector) must be performed non-speculatively.  And, for both scalar and vector load instructions, part of the memory being accessed may be idempotent and part may be non-idempotent.  In which case the former parts may be accessed speculatively, but the latter parts must only be accessed non-speculatively.

This also means that one shouldn't load an inactive non-idempotent element (since even non-speculatively it is not intended to be accessed).  Although if one is loading a nearby active non-idempotent element, one might still wonder whether one could do an access within that non-idempotent region that is larger than the active non-idempotent element (irrespective, actually, of whether there are other nearby active or inactive elements).

The issue with doing that, typically, will be that non-idempotent regions will not support large/wide accesses (since, for example, these contain memory-mapped registers) and will raise an Access Fault on an attempted access larger than what is supported (which typically may be just 4B or 8B).

But even if one knows that the non-idempotent region supports wide multi-element vector accesses, one still has to worry about what possible side effects may result from accessing inactive elements.  Essentially the software needs to know the specific characteristics of the non-idempotent region that it is trying to perform a vector load to that has inactive elements - and whether it is ok or safe to do what it is trying to do.

closed 2025-07-16 by Qin0803