riscv/riscv-isa-manual #2145: Add check if normative rule text in manual is being changed

james-ball-qualcomm, 2025-07-19:
Now that we have started tagging normative rule text snippets in the ISA manuals for the CSC certification test plans (and probably other uses in the future), we'll need a facility to detect when someone attempts to make changes to these tagged portions of the ISA manuals. The tagging is very visible in the adoc source files but someone could still inadvertently change the text not realizing the implications of their actions (i.e., breaking the CSC test plans).

I'm going to create a Ruby gem that we can use in this repository to perform an intelligent compare of the tagged normative rules against a known reference. If there is a difference, we could either trigger a pre-commit failure (like UDB does) and/or add a GitHub action that causes the PR to fail a check.

Ideally the reference is only allowed to be changed by select individuals including those responsible for CSC test plans so that a user doesn't just change the reference file to work around the miscompare.

Adding interested parties: @Timmmm, @dhower-qc, @jrgrimm11, @wmat, @kbroch-rivosinc

james-ball-qualcomm, 2025-07-21:
Let me clarify. Once text in the ISA manual is tagged to be used in a normative rule (i.e., all of the tagged text with a id prefix of "norm:"), we'll have CSC test plans that include the anchored text (in its list of normative rules) and we'll have tests that match the behavior of the tagged text. What happens if someone changes the tagged text? If it is a non-normative change (i.e., doesn't change the behavior) such as formatting, spelling, better wording then this doesn't create a problem for the tests and test plans. However, if it is a normative change that changes the behavior then making such a change will break the test plans and tests.

For example, say there is tagged text that says "RISC-V has 3 privilege modes: U, S, and M". And say someone edits this to say "RISC-V has 3 privilege modes: User, Supervisor, and Machine". That is a non-normative change to the tagged text. However, if RVI decides to extend RISC-V to add another privilege mode and changes the text to say "RISC-V has 4 privilege modes: U, S, M, and X". This could break the CSC test plans and tests. Or you can imagine other text in the ISA manual that is tagged for inclusion in normative rules is changed and could break the CSC test plans and tests.

To avoid breaking the CSC test plans and tests with ISA manual changes, my suggestion is that the user get prevented from either committing or having their PR accepted if they are attempting to change normative rule tagged text ("norm:" prefix). They would need additional review from someone knowledgeable about the CSC test plans before their commit/PR is accepted.

Here's the details of my proposal:

1. Create a smart diff utility that ignores formatting changes (e.g., whitespace)
2. Run the smart diff utility when ever a user builds any targets in the Makefile and give them a warning if the smart diff utility detects a change to tagged normative rule text.
3. Run a GitHub action when a PR is created that won't allow the PR until the diff is resolved (somehow).

james-ball-qualcomm, 2025-07-21:
Backing away from my proposal after meeting with Derek and Tim. We can always add what I proposed later if it becomes a problem that people are changing tagged normative rule text without considering or being aware of the impact on the CSC's test plans and tests. Instead, we could just provide a warning and summary of changes (including deletions) whenever they build the manual. This means calling a smart diff tool in the Makefile. It would probably also make sense to have a GitHub action run the script (as informative, not blocking a PR merge)

wmat, 2025-07-21:
One thing I'll throw out there is possibly maintaining a branch of
riscv-isa-manual that you use for UDB purposes and sync it to the main
branch daily perhaps.  This will protect your usage of the main branch of
the riscv-isa-manual from breaking either the current build functionality
and your use of the src adoc files in UDB.

Maybe I'm worrying too much though.

james-ball-qualcomm, 2025-07-24:
In my ISA manual fork, I've changed the Makefile to generate the normative rule tags whenever a target (HTML, PDF, or EPUB) is built. This is a stepping stone to work towards this GitHub issue. Next step will require my intelligent compare utility (will be a Ruby gem) to compare against the expected normative rule tags and generate a nice warning of differences.

james-ball-qualcomm, 2025-12-09:
Can you please add a link/reference to that thread? I'm out of the loop.