riscv/riscv-isa-manual #2148: Can IFU fetch instruction with PMA=IO ?

rock-ifly, 2025-07-21:
**_Can IFU fetch instruction with PMA=IO ?_**

1)  RV spec says IO region can be configured with pte_x=1, which means that ifu can fetch instructions with pma=IO

> I/O regions can specify which combinations of read, write, or execute accesses to which data widths
are supported.

> For systems with page-based virtual memory, I/O and memory regions can specify which combinations

of hardware page-table reads and hardware page-table writes are supported.

2) RV specs also says non-idempotent( IO region used to be non-idempotent) memory shouldn't support speculative access.
    Usually, bpu works at the beginning of pc generating, without pma information. This require that the instrution fetch must NOT be in the IO region.
    That is to says, harts with bpu should make sure all the codes must be with pma=NC or Cacheable, and fetch instuction from IO region shoule be WRONG.  --> **pages with PMA=IO + pte_x=1 should generate error ?**

> While hardware should always be designed to avoid speculative or redundant
accesses to memory regions marked as non-idempotent, it is also necessary to
ensure software or compiler optimizations do not generate spurious accesses to non-
idempotent memory regions.

gfavor, 2025-07-21:
Yes, one can fetch from an I/O PMA region.  See the last paragraph of section 3.6.7 in the Priv spec - which addresses the specifics of this question.

gfavor, 2025-07-24:
Closing since there has been no further posts and I believe the questions have effectively been addressed.

closed 2025-07-24 by gfavor