riscv/riscv-isa-manual #2159: Should cross page parts raise va_range_fault?

AlexGJL, 2025-07-25:
Hi, for a load/store instructions that split to two parts due to access address size  across page size boundaries. the virtual address of first part is right, but the virtual address for second part : va[64:39] != va[38] for sv39. should raise a page fault because of the va range fault.? and what address should write to *tval for cross page instructions？

gfavor, 2025-07-25:
The spec, for Sv39, says "Instruction fetch addresses and load and store effective addresses, which are 64 bits, must have bits 63–39 all equal to bit 38, or else a page-fault exception will occur."  In this case there are two page addresses to be translated and the second one violates the preceding statement - and hence should raise a page fault.

*tval is written, as the spec says, with the "faulting virtual address". In this example, the VA of the faulting second part is what should be written into *tval.  (*tval is intended to tell software the address that is causing the problem.  Writing the first part's VA into *tval would be useless and misleading to a page fault handler.)

allenjbaum, 2025-07-28:
> No, all bytes must be individually be accessible for the access to succeed
> overall.

Not sure I understand that short answer.
I would have answered something like "Yes but, all bytes must be
individually be accessible for the access to succeed overall."
And that is possible in this case (though it's an unlikely corner case)

So,  pedantic/on:
If the sign extended address is negative, such that the address "overflows"
from a negative address to a positive address (in the range 0..7 most
likely),
and VA=0 has a legal /accessible mapping, then HW setting 63:39 to all
zeroes seems perfectly legal - it's what would happen naturally if you just
incremented the address (no special HW required)
Note that post page translation(s), those 2 pages' PA's could be
contiguous, or may not be.

PMP restrictions may still doom that if the two halves are treated as a
single access and the 2 translated PAs aren't covered by a single PMP entry
If split into two transactions, each could succeed even if covered by two
different PMP entries.
If the initial address is positive, though, then crossing the boundary
yields a still-positive, non-canonical (illegal) address, and then clearing
bit 39 would not be legal.

pedantic/off

What am I misunderstanding here?

allenjbaum, 2025-07-28:
> Wrapping around from negative addresses to positive addresses is fine, but
> it isn't relevant to the question at hand. The question poses that bits
> 63..39 are not all equal to bit 38 for a subset of the access. This can
> only happen if the base address is just below 2^38. It can't happen if the
> address wraps around from negative to positive addresses, because in this
> case, for each byte address, bits 63..39 are equal to bit 38.

OK, we agree on the result, but I didn't interpret the question the way you
did. That's my superpower, as I've explained.
I thought he was specifically talking about the wraparound case; I'm not
sure how that can happen otherwise. I should reread what he sent.

ved-rivos, 2025-07-28:
Consider the valid address ranges for Sv48. 
```
0x00000000_00000000 - 0x007FFFFF_FFFFFFFF
0xFF800000_00000000 - 0xFFFFFFFF_FFFFFFFF      
```         
If instruction is straddling 0xFFFFFFFF_FFFFFFFF and 0x00000000_00000000 then both are valid addresses. One has to have an instruction straddling 0x007FFFFF_FFFFFFFF and 0x00800000_00000000 or 0xFF7FFFFF_FFFFFFFF and 0xFF800000_00000000 to have one invalid and one valid address for the fetch.

aswaterman, 2025-07-28:
> OK, we agree on the result, but I didn't interpret the question the way you did.

Gotcha.  As you say, we agree on the semantics.

gfavor, 2025-07-30:
Closing since I believe all questions have been answered.