riscv/riscv-isa-manual #2172: PTE.N in Svnapot spec

Chen-rui55, 2025-07-31:
Hi, I have some questions regarding Svnapot in the spec. As described in Chapter 13. "Svnapot" Extension:

> In Sv39, Sv48, and Sv57, when a PTE has N=1, the PTE represents a translation that is part of a range of contiguous virtual-to-physical translations with the same values for PTE bits 5–0.

1. When a PTE has N=1, does this only apply to leaf PTEs? In other words, are only leaf PTEs permitted to carry N=1, while intermediate/non-leaf PTEs must not have N=1 set?

> Implicit reads of NAPOT page table entries may create address-translation cache entries mapping a + j×PTESIZE to a copy of pte in which pte.ppn[i][pte.napot_bits-1:0] is replaced by vpn[i][pte.napot_bits-1:0], for any or all j such that j >> napot_bits = vpn[i] >> napot_bits,

2. In the expression pte.ppn[i][pte.napot_bits-1:0], does the index i refer to each level in the page table hierarchy?
For example, in Sv39 mode:
- When processing pte.napot_bits, do we need to right-shift all VPN segments (i.e., vpn[2] >> napot_bits, vpn[1] >> napot_bits, vpn[0] >> napot_bits)?
- Or is it only necessary to shift the last-level VPN (vpn[0] >> napot_bits)?

aswaterman, 2025-07-31:
1. Correct.  The spec doesn't explicitly say this, but it is implied by the wording "the PTE represents a translation".  Non-leaf PTEs don't represent translations.

2. Refer to the leftmost column of the table.  The only defined NAPOT regime (64 KiB) has i=0.  With that said, there is no shifting involved at all.  The translated physical address is formed by replacing ppn[0][3:0] with vpn[0][3:0].

Chen-rui55, 2025-08-01:
Got it, thank you!

closed 2025-08-01 by aswaterman