riscv/riscv-isa-manual #2174: On Vector Mixed-Width Issues

wusong-ai, 2025-07-31:
Here are three questions, thanks for the reply.

Regarding mixed-width instructions:
1、Must the SEW/LMUL ratio restrictions shown in Section 31.4.4 of the RISC-V Manual be strictly adhered to (as presented in the table below)?
[image]
2、Should an illegal instruction exception be raised if ratios outside this table are encountered?

Additionally, for vector load/store operations (excluding whole register and mask operations):
3、Destination/source register configurations require calculation using the width (EEW) field in the instruction code alongside the SEW and LMUL values in the vtype register. Should this configuration be considered a mixed-width instruction as described in this section?

wusong-ai, 2025-08-01:
Our primary concern pertains to the vector load/store instructions: is it permissible for the ratio sew/lmul to exceed 64? Our analysis reveals a specific scenario under vlen = 128: when lmul = 1/8 and sew = 16, an eew of 8 (i.e., width[2:0]=000) results in a calculated emul of 1/16, which constitutes an invalid value. Conversely, with the same lmul = 1/8 and sew = 16, but an eew of 32 (i.e., width[2:0]=110), the calculated emul is 1/4, a valid value.

This raises a critical implementation question for mixed-width operations: should all scenarios where sew/lmul > 64 be universally deemed invalid, or should invalidity be determined solely when the resulting emul calculation yields an illegal value?

aswaterman, 2025-08-01:
The combinations you mentioned are not mandatory to support under the rules I mentioned, and so you are free to trap them (or set vill in the case of a setvl instruction).

In general, any combination with SEW/LMUL > EMUL falls outside the mandatory set, so setting vill when attempting to select such a combination with a setvl instruction is a legitimate choice.

wusong-ai, 2025-08-01:
Thank you again for your reply. However, I'm not sure about the part SEW/LMUL > EMUL mentioned in your last response. Could this be a typo?

wusong-ai, 2025-08-05:
Thank you, my problem is solved!

closed 2025-08-05 by gfavor